DSheirer/sdrtrunk #26: RTL Tuner - Cannot disable Auto Frequency

uj, 2015-09-19:
Even if I uncheck Automatic Frequency Control under Decode, and set the limit to 0 under Decode, and set PPM to 0 under the Tuner, it still jumps the frequency up to a slightly higher frequency when I click Save on the channel.  I have to then manually change it back to the proper frequency.  This happens on the official 0.1.5 version, and also on (as of Sept-19-2015) github version if I build it myself.

DSheirer, 2015-09-21:
In order to avoid the DC spike that is present in most of the tuners, the tuner controller will automatically select a center tuned frequency that places your requested channel somewhere to the left or right of the center frequency and within the tunable bandwidth.

AFC controls fine tuning discrepancies between your requested channel frequency and the demodulated carrier frequency, when that carrier is within +/- of the AFC frequency maximum selected.

Do you need the center tuned frequency to be the same as the requested channel frequency?

uj, 2015-09-22:
I thought the center tuned frequency needed to be the same as the requested channel frequency, or at least within the blue strip that says "NBFM".  I thought that blue strip was the entire width that the radio could see.  It is tuning the center outside of the blue strip range.  Maybe I am misunderstanding how it works.  With Unitrunker, it keeps the center on the actual frequency.

DSheirer, 2015-09-22:
The blue strip will be the tuned bandwidth that is being processed by the channel.  You can have as many channels as will fit within the visible spectral display, so long as each channel fits within the bandwidth. 

If you're using the NBFM decoder to listen to a channel ... start the channel running and then manually adjust the tuner's center frequency to be the same as the channel's frequency while listening to the audio ... the audio should sound more distorted when your channel (blue strip) is overlapping the center DC spike.

I suspect that if you're not seeing the center tuned frequency in unitrunker, that it may be hiding the actual center tuned frequency from the user, but I don't know for sure.

uj, 2015-09-22:
Ah, that makes sense.  Thank you for clarifying that.

closed 2015-09-26 by DSheirer