DSheirer/sdrtrunk #30: Audio Streaming Support

DSheirer, 2015-10-03:
Add support for Shoutcast/Icecast network streaming
-Multiple, simultaneous audio streams
-Stream designation by Alias/Talkgroup

imagesafari, 2016-08-19:
This software is awesome. Thanks so much for your effort. With Audio streaming, it'd be cool to have a web interface to the recordings too -- like Trunking Recorder + Unitrunker. Happy to provide a demo link if you haven't seen it. The data is all there, just needs a web interface.

DSheirer, 2016-08-19:
Thanks!  Yes, if you can post a link, I'll give it a look.

imagesafari, 2016-08-19:
Link here: http://www.scannerbox.us/TrunkingRecorder/

I can give you a link to a running web interface so you see what it does via PM

imagesafari, 2016-09-06:
I can't figure out how to send a PM on github.. drop me an email and I'll shoot you a link to a working Trunking Recorder. I've got some other ideas that I'd love to share / help with if I can. 

jon@anhold.com

closed 2017-01-07 by DSheirer