DSheirer/sdrtrunk #35: Create AM Squelch Control

DSheirer, 2015-11-29:
Create a squelch control for the AM decoder

ah1102, 2020-04-28:
I would also like to see in the program noise reduction for analog FM and AM. I also noticed an echo in NFM.

The program may be more universal. With hackrf, it would be possible to have multichannel reception on the air or amateur radio.

ah1102, 2023-01-25:
Dear developer! Enter squelch for AM. It is not difficult to add it for AM, taking it from the NBFM mode. Just add a similar feature for AM.

DSheirer, 2023-05-02:
AM squelch control implemented in branch.  AM & FM now share common squelch control code.  Need to add dynamic channel noise floor monitoring to auto-adjust squelch threshold and change the squelch threshold to be a value that's added to the current noise floor.
[image]

DSheirer, 2023-05-07:
Closed via PR.

closed 2023-05-07 by DSheirer