DSheirer/sdrtrunk #36: Create FM squelch control

DSheirer, 2015-11-29:
Create an FM squelch control and add it to the NBFM decoder

ahoy, 2018-03-15:
@DSheirer Do you have a high level idea of how to accomplish this? I would like to take a crack at implementing this using Gnuradio's simple_squelch method.

DSheirer, 2018-03-17:
I've been on travel the last 2 weeks, sorry for the delayed response.

Some background .... the ChannelState class (and any subclass implementations) controls the squelch state for audio to be distributed.  It receives DecoderStateEvent instances from any of the DecoderState or Decoder component instances.

The DecoderFactory.getPrimaryModules() method creates the modules that make up the FM decoder.  

The FMDemodulatorModule receives complex sample buffers, performs baseband (IQ) filtering and FM demodulation and produces unfiltered real sample buffers.  The 

The DemodulatedAudioFilterModule performs filtering on the unfiltered real sample buffers to condition the audio for the audio module.

The AudioModule listens for SquelchState signals from the DecoderState and receives demodulated real sample buffers and converts them to audio buffers.

The AlwaysUnsquelchedDecoderState controls audio squelching by essentially setting to it always unsquelched.

(end background)

I'm assuming that you're going to be monitoring the incoming complex sample buffers to detect the presence of a modulated signal.  I'd recommend:

Create a new FMSquelch class that extends Module and implements both IComplexBufferListener and IDecoderStateProvider interfaces.  The IComplexBufferListener interface allows you to receive complex baseband samples for the channel so that you can monitor for signal in each buffer.  The IDecoderStateProvider interface allows you to broadcast DecoderStateEvents to the processing chain to command the ChannelState to (un)squelch the audio output for the channel.

Another (preferred) option is to extend the FMDemodulatorModule to add the FMSquelch functionality.  This would allow you to control which baseband buffers get demodulated and passed on to the rest of the chain.  Otherwise, you'd simply be turning the audio playback squelch on and off, but otherwise all baseband buffers would still be demodulated.

You'll have to issue an initial DecoderStateEvent using the Event.START and then follow that up with a continuous stream of Event.CONTINUE and finally an Event.END.  The ChannelState receives these events and controls the overall squelch state for the channel.  If the ChannelState doesn't receive a a DecoderStateEvent within 1200 milliseconds, it will change the channel state to FADE and then in 2000 milliseconds, it will reset the ChannelState to IDLE.  Therefore, you need to broadcast a DecoderStateEvent at least every <1200 milliseconds to maintain the squelch state.  Once you send the START event, you can probably send a CONTINUE event for each buffer that contains a modulated signal, then issue an END event on the first buffer that doesn't contain a signal.

The FM demodulator currently uses the AlwaysUnsquelchedDecoderState, but you can create a new class that extends DecoderState that doesn't automatically set the squelch on startup.  If you look at the AlwaysUnsquelchedDecoderState, you can copy it and simply leave out the part where it broadcasts the ALWAYS_UNSQUELCHED event.

Update the DecoderFactory to use your new Squelch class and the new DecoderState class for the NBFM decoder module stack.

DSheirer, 2018-03-17:
An additional comment ... in version 0.4.0 I'm changing the channel sample rate from a default of 48kHz to a variable sample rate between 25.0 - 50.0 kHz.  Normally this will be a 12.5 kHz filtered channel that has 2x oversampling to 25.0 kHz.  But, depending on the overall sample rate of the tuner, I may not be able to get an even/clean 12.5 kHz channelization, and therefore the channel sample rate could go to just under 50.0 kHz.

ahoy, 2018-07-27:
Thanks for all the info. I've been (slowly) trying out a couple different approaches to this, but I can't seem to get anything reliable. I really like the idea of "short circuting" the signal path to keep squelched packets from getting any farther down the processing chain.
I was digging around in the gnuradio source and found that they use "tagging," which allows metadata to be attached to the raw packets. Tagging packets of IQ with signal strength can be used early on, as well as extracted by the UI to display some sort of an S meter. This would be handy for squelch adjustment.

Slightly off topic... is this an appropriate place to discuss the development of this?

Thanks and more thanks for this great project!

DSheirer, 2018-07-31:
No worries ... if it relates to the FM squelch issue, it's on topic or you can email me directly.

Handwrit, 2019-01-15:
Because I live in a rural area, we have a mix of non-trunked P25 and FM. The police use P25, but fire/ambulance uses FMN. It would be nice to be able to use just one program.

Any idea how solvable FM squelch is? It looks like something as simple as "if this area of the waterfall turns red enough, then unsquelch" would work alright.

DSheirer, 2019-01-15:
It isn't too hard to implement ... just requires time.  I'll try to get this added soon.

Handwrit, 2019-01-15:
I'm very appreciative! I'm also well aware that the name "SDRTrunk" pretty much implies mine is an edge case.

DaveNF2G, 2019-11-27:
Curious about progress on this.  I get a lot of junk decodes in LTR-Passport because the program is trying to read the noise.  

I also like the QCII idea and would like to suggest GE*Star as well.

ah1102, 2020-04-28:
I would also like to see in the program noise reduction for analog FM and AM. I also noticed an echo in NFM.

The program may be more universal. With hackrf, it would be possible to have multichannel reception on the air or amateur radio.

BloodBlight, 2020-09-28:
I noticed that this is on the list for both milestone 0.5 and 0.6.  This would be supper helpful I think.  I am trying to monitor the Marine bands, and there are a lot of them.  :)

ivanpiter, 2021-01-05:
Dear  Denny!

Please, add squelch for NBFM and AM demodulators because they are useless for me without it.
It s mixed analog and digital system around me and without squelch I can not use your perfect software effectively.
Here is my typical mixed radio frequency environment:
[image]

> 
> It isn't too hard to implement ... just requires time. I'll try to get this added soon.

PR #1100 (2021-11-14): #36 NBFM squelch control. Updates NBFM decoder to incorporate squelch…
#36 NBFM squelch control. 
* Updates NBFM decoder to incorporate squelch control.  
* Squelch is a configuration item and also available to adjust at runtime with new 'Channel' tab in Now Playing window.  
* All decoders were updated to produce continuous channel power (dB) measurements that are displayed in the Channel tab's new Power Meter.
* Talkgroup identifier can be specified in decoder config.  Value is assigned to all audio segments produced to enable recording, streaming, etc.
* Playlist editor updated to include n [... cut]
closed 2021-11-14 by DSheirer