DSheirer/sdrtrunk #40: Configuration option to display IDs in decimal rather than hexadecimal

davidmckenzie, 2015-12-22:
Super low priority - just something for the "nice to have" list. Would be to good to be able to display talkgroup and unit IDs as decimal in the event log. Perhaps through right clicking anywhere in the log.

DSheirer, 2015-12-23:
This would impact aliases and alias IDs.  Do you envision this being an either/or switch, so that the system uses all decimal values and users would setup their aliases with decimal values, and vice-versa for hexadecimal?

davidmckenzie, 2015-12-23:
I was thinking just as a purely visual thing, nothing changing underneath, just choose whether to display the event log in dec or hex. Might it be easier to just display both values, e.g. "7730 (30512)"?

vsboost, 2017-09-04:
Would love to see this as well.

closed 2018-12-24 by DSheirer