DSheirer/sdrtrunk #45: Unable to decode San Francisco Motorola Type II SmartZone / P25

gbin, 2016-01-16:
I took a raw recording around 853MHz to give you a better idea [here](https://drive.google.com/a/gootz.net/file/d/0B9tDJ_NcZuDtemZBcEluMUYyNlk/view?usp=sharing).
The frequencies are documented [here](http://www.radioreference.com/apps/db/?sid=60) and they seem to match.
With gqrx I can get the analog bits in narrow FM without any problem but I tried to decode the digital part with sdrtrunk under Linux Arch and so far I did not even get one event logged :(

Questions: am I supposed to tune in to the control channels (852.86250c, 852.36250c etc...) or the voice ones ? Is it possible to follow a specific talkgroup ?

DSheirer, 2016-01-17:
sdrtrunk doesn't yet have a decoder for Type II control channels (Feature/Issue #2), so it won't be able to do trunked traffic channel following.

If the voice channels are using P25 CAI, then you can setup the decoder with P25/C4FM for each frequency and decode them that way.  I don't have any SmartZone/P25 systems around here to test against.

I have 1 Type II Moto 3600 trunked system nearby, but they are using analog voice channels.

gbin, 2016-01-17:
Thanks for the clarification, if you want some long recordings of the spectrum to debug/test the app with a Type II channel, I'd love to send you those (my second RTL is coming so I can get a full 2Mhz bandwitdh range) !

DSheirer, 2016-01-27:
Do you know if this system is using VSELP vocoders or the IMBE vocoder?  VSELP is an older proprietary Motorola format and IMBE is the standard P25 codec.

DSheirer, 2016-01-28:
I added preliminary support for VSELP audio codec voice frames VSELP1 and VSELP2 based on a document that I had.  Upon confirmation that VSELP is being used on the San Fran system, maybe we can find the full format and I can add VSELP codec support to the JMBE library.

If VSELP is in use, you'll see the decoded messages in the messages tab.  They simply say VSELP1 Audio Message and the binary data.

wingo395, 2017-08-05:
I think I'm in the same boat with Marin county, just north of San Francisco.
http://www.radioreference.com/apps/db/?sid=1316
Motorola Type II SmartZone - Analog and APCO-25 Common Air Interface
( can you tell I'm new to all this?  :) )

omgitsgela, 2018-03-13:
Is there any way we can help with this issue? We have a SmartZone / P25 II system statewide here in Massachusetts as our main communications system for most services. It would be great to get this working with sdrtrunk!

JoqOfSpades, 2020-02-01:
> I have 1 Type II Moto 3600 trunked system nearby, but they are using analog voice channels.

If you need someone to help you test it, I'm in the Raleigh, NC area that has a hybrid P25/Analog Type II SmartZone, and currently have to use a discriminator tapped RadioShack trunking scanner (PRO-94) into DSD running on a Raspberry Pi.. Not a very elegant solution, but it works. I would love to have an SDR solution!

ryanthomp, 2020-04-30:
Having the same issue in my area, I used the RR interface to add an MOT in my area and it added all the individual frequencies and tuned into them but even when they were clearly transmitting SDRTrunk picked nothing up, [https://www.radioreference.com/apps/db/?sid=3067](url). This is a proprietary system I believe.

Nokoa, 2020-11-25:
Following up on this, I also would like to use this for the SF Motorola Type II SmartZone system, most voice channels on it are analog. Was this feature ever added?

BrendanMonaghan, 2020-11-30:
> Following up on this, I also would like to use this for the SF Motorola Type II SmartZone system, most voice channels on it are analog. Was this feature ever added?

Also very curious about SF Motorola Type II. Just starting out, but willing to test solutions if that's helpful.

DSheirer, 2023-11-14:
Closing as duplicate to #2

closed 2023-11-14 by DSheirer