DSheirer/sdrtrunk #50: Feature: Talkgroup Hold

DSheirer, 2016-02-15:
Add ability to hold on a specific talkgroup.  

When monitoring a system and a specific talkgroup is active, add ability to hold (only listen to) that specific talkgroup, until the hold is removed.

DSheirer, 2019-08-29:
This feature is still open and hasn't been implemented yet.

malchiuri, 2020-04-11:
Hello,
Your software is beautiful. Working fine all the time.
Talkgroup HOLD is important to remain on specific talkgroup on sudden event.
Is an "HOLD" button can make a temporary list with that specific talkgroup number, have Audio Priority = 1, and at the same time create blacklist with same list name, with Audio Priority = Do Not Monitor, to override and silence the other lists, until you release and let it run normal again.
It will help simplify monitoring without creating, arranging and folowing too many lists .
Thank you.

FrostTheFox, 2020-12-20:
Just want to drop by with a +1 on this issue. I've converted at least a couple different people to SDRTrunk and while I personally host a calls node and don't have a need for this, the people I've converted are more used to traditional scanners. SDRTrunk's alias system is very powerful but it would be great to have this as an option for people who just want to casually hop between listening to a single talkgroup.

PaulWeiss, 2022-08-22:
SDRTrunk is amazing software! Works as well, sometimes better than my Uniden SDS100 and Whistler TRX-1. SDRTrunk is easier to configure too!

+1 to add a hold talkgroup feature. This would make this software 100% complete in my view.

Hermite123456, 2022-08-24:
> SDRTrunk is amazing software! Works as well, sometimes better than my Uniden SDS100 and Whistler TRX-1. SDRTrunk is easier to configure too!
> 
> +1 to add a hold talkgroup feature. This would make this software 100% complete in my view.

Totally agree. This would make the app pretty complete :)