DSheirer/sdrtrunk #52: Feature: SDRPlay RSP1 Tuner Support

DSheirer, 2016-02-15:
Add support for RSP1 tuner

n2cr, 2017-01-29:
Suggest that this enhancement be modified to add support for both SDRPlay RSP2 as well as RSP1.

infecticide, 2018-07-03:
I've just recently purchased an RSP2Pro.    Is there anything a non-programmer can do to assist with this feature request?

ImagoTrigger, 2020-02-02:
bump: https://github.com/SDRplay/examples/blob/master/play_sdr.c

StaRipper, 2020-06-01:
Bump the beat

LordDonkey, 2020-06-02:
I second this. I've been eyeing up the MSI.SDR. It has up to 10 MHz of bandwidth and would cover enough spectrum to fully cover the system I monitor. It uses the same drivers as the RSP1.

StaRipper, 2020-06-02:
> 
> 
> Add support for RSP1 and RSP2 tuner

On v0.5.0 Alpha1  not recognizing Sdrplay RSP1

StaRipper, 2020-07-06:
[https://forums.radioreference.com/threads/dsdplus-2-256-with-sdrplay-rsp1.409013/](url)

NoahJaehnert, 2020-08-02:
Would love to see SDRPlay support added as well. I have an RSP2 that I'd like to use with sdrtrunk.

dseven, 2020-08-02:
Came here trying to find out if I could use my RSPdx. The P25 system that I monitor spans about 4.5MHz - more than one R820T can cover without retuning, etc..... so add me to the wish list!

liquidmayo, 2020-10-29:
adding another rsp1a to the list

TheButterZone, 2021-06-02:
Add another RSP1.

phipac, 2021-09-23:
For me, having RSP support would make this already fantastic package even more so!

patomoi, 2022-03-24:
would be great that this software supports RSP family

PR #1414 (2023-01-15): #52 Add sdrplay tuner support.
Closes #52.  Adds support for SDRplay RSP tuners.
PR #1453 (2023-02-19): #52 SDRPlay RSP tuner support.
Closes #52.

Adds support for all RSP tuners
closed 2023-02-19 by DSheirer