DSheirer/sdrtrunk #55: Create Windows and Linux Installers

DSheirer, 2016-02-20:
Create installer that:

a) Installs latest version of java 
b) Sets up the path environment variable for java
c) Installs start menu item

Retain the batch scripts for starting the program.

ImagoTrigger, 2017-02-11:
I will set you up a bitten build slave that poops out a new windows installer every time you make a commit to master.

ImagoTrigger, 2017-02-15:
[image]

getting there... now to run it thru makensis and post back the .exe

chudgoo, 2017-02-15:
Does it maintain the configs and playlist between updates?

ImagoTrigger, 2017-02-17:
I don't see why it wouldn't.  those files won't be part of the installer.

ImagoTrigger, 2017-03-01:
here is a prototype windows installer with jre checker, installer, launcher/shortcuts and uninstaller.  You can try it here: https://github.com/ImagoTrigger/sdrtrunk/releases/download/v0.3/SDRTrunk-install.exe

[sdrtrunk-nsis-prototype-src.zip](https://github.com/DSheirer/sdrtrunk/files/811945/sdrtrunk-nsis-prototype-src.zip)

chudgoo, 2017-03-01:
Nice! I will test this now on the windows VM.
I took a stab at a mac/linux version of something similar after realizing I was repeating the same steps every couple of days after a new commit. 

I'm not a coder and suck at bash scripting for the most part, but here's what I came up with...
The only prerequisite is that you have your known good/production playlist, images, etc directories in a common spot, I guess....   Oh, and I gave up on doing the java and ant version check and upgrade part because work was making me work...

!/bin/bash
#Back up old install (excluding recordings)
zip -r sdrtrunk-lastversion-backup.zip ~/sdrtrunk -x *recordings*
#Unshit the bed if last run screwed up and left things in a weird state...
rm -rf ~/sdrtrunk
rm -rf ~/tempsdrtrunk
rm -rf ~/JMBE
#Get latest version of SDRTrunk and build/install.
mkdir ~/tempsdrtrunk
cd ~/tempsdrtrunk
git clone https://github.com/DSheirer/sdrtrunk
cd ./sdrtrunk/build/
ant
unzip -o ../product/sdrtrunk*.zip -d ~/
cd ~
rm -rf ~/tempsdrtrunk/
#Copy config files from local backup into new install dir.
yes | cp -rv ~/SDRTRUNK-CONFIG-FILES/* ~/sdrtrunk
#Get latest version of JMBE and copy to sdtrunk dir
git clone https://github.com/DSheirer/JMBE
cd ~/JMBE/jmbe/build/
ant
yes | cp ../library/jmbe-0*.jar ~/sdrtrunk
cd ~
rm -rf ~/JMBE/
#Make it so.
bash ~/sdrtrunk/run_sdrtrunk_linux.sh

davidmckenzie, 2017-03-01:
You actually don't need to copy the configs and do some scary rms each time - if you just extract the tarball over the existing directory it will update the changed files and preserve your configs. I'll have a look at writing up a bash script for upgrades soonish. :)

chudgoo, 2017-03-01:
Awesome!  I think not having a simple installer has been the number one set back in trying to get friends in different parts of the country to give it a go.  Meanwhile they're struggling with the cluster that is UniTrunker > Virtual Audio Cable > DSD+ > BUTT or something similar.   (Nothing against the authors of any of those applications, but SDRTrunk fits my needs much better and provides a more elegant and straightforward approach. Really loving this software... )

Also, I tried running the installer @ImagoTrigger posted and it worked without a hitch!
The only issue I'm running into on this particular VM is some annoying proxy stuff that prevents me from downloading the ZADIG stuff with the sdrsharp-x86 installer script.  
Basically SDRTrunk installs and runs perfectly... I just don't have any tuners listed.  :)
Will try again from home later...

davidmckenzie, 2017-03-01:
Ok, here's an upgrade script - written on OSX, haven't tested on Linux. Pretty limited, but might help :)
````
#!/bin/bash

#############################################################
#
# SDRTrunk Update Script
# ----------------------
# Retrieves the latest copy of source, compiles, and updates
# Assumes install directory is ~/sdrtrunk
#
# TODO:
#  * Cleaner backup
#  * Interactive prompting for install dir
#  * First time installation including jmbe
#  * Make the script suck less
#
#############################################################

# Is SDRTrunk running?
if pgrep -f SDRTrunk > /dev/null
then
	echo "SDRTrunk is running, please kill it with fire"
	exit 2
else
	echo "SDRTrunk not running, proceeding"
fi

# Make a backup
cp -pRv ~/sdrtrunk ~/sdrtrunk-backup

if [ -d ~/sdrtrunk-source ]
then
	echo "Source directory exists, proceeding"
else
	echo "No source directory, cloning"
	git clone https://github.com/DSheirer/sdrtrunk.git ~/sdrtrunk-source
fi

cd ~/sdrtrunk-source
git pull
cd build
echo "Compiling SDRTrunk"
ant
# Did the build succeed?
if [ $? -eq 0 ]
then
	echo "Compile complete!"
else
	echo "Uh-oh, compile returned $?"
	exit 2
fi

echo "Extracting files"
cd ~/
tar -xzf ~/sdrtrunk-source/product/*.tar.gz
echo "Done!"
````

ImagoTrigger, 2017-03-02:
cool forgot about zdiag driver i can look into getting that incorporated. as for repeating steps, i plan on it

ImagoTrigger, 2017-03-05:
zdiag is in.  installer should automatically get created now each time denny commits.

https://github.com/ImagoTrigger/sdrtrunk/releases

chudgoo, 2017-03-06:
@ImagoTrigger 
I tried it on the same proxy'd VM and it worked flawlessly.  
Was a bit panicked when I saw all of my aliases missing, but quickly remembered I was in the VM.  ;)

chudgoo, 2017-03-06:
@davidmckenzie 
I also tried your upgrade script.  Rock solid on Mac.
I was wondering if you had the magic to do a java / ant version check and upgrade or install if necessary. Would probably depend on Homebrew, which installs easily enough with:

/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"

davidmckenzie, 2017-03-06:
Hmmm yeah I guess that would depend on how you installed it, since there's a few different package managers to use for ant. I have a vague recollection that installing homebrew alongside macports causes bad things to happen.

For an install script, perhaps it would be best to simply check if ant exists, and if not link to the documentation page on how to install it.

chudgoo, 2017-03-06:
@davidmckenzie  Yeah I seem to recall struggling a bit to get Ant installed and configured correctly. I wonder what the most "K.I.S.S." method of an all in one installer for mac/linux might be.

ImagoTrigger, 2017-03-25:
emails are being sent out when a build completed.  is there anywhere i should have it send (besides me).  it looks like this:

> Subject: SDRTrunk build completed!
Date: Thu, 23 Mar 2017 21:59:56 +0000
X-Mailer: sendEmail-1.56
MIME-Version: 1.0
Content-Type: multipart/related; boundary="----MIME delimiter for sendEmail-227813.720703125"

------MIME delimiter for sendEmail-227813.720703125
Content-Type: text/plain;
        charset="iso-8859-1"
Content-Transfer-Encoding: 7bit

Completed build # 20 for 0.3.0--beta9 7a0b377971fb6bb80a4496034a22f8d1b7cb9021

ftp://anoka.radiowing.com:2121/SDRTrunk_b20_7a0b377971fb6bb80a4496034a22f8d1b7cb9021.exe

https://github.com/ImagoTrigger/sdrtrunk/releases

------MIME delimiter for sendEmail-227813.720703125--

closed 2022-11-26 by DSheirer