DSheirer/sdrtrunk #61: P25 Decoder - rolling control channel following

DSheirer, 2016-03-10:
Some P25 systems rotate control channel frequencies across the set of allocated channels.  Add the ability to specify a list of control channel frequencies and have the decoder switch among the frequencies until it lands upon a control channel.  Once the decoder stops decoding a P25 control channel for a user specified timeout period, switch back to a hunt mode.

N3FM, 2018-02-02:
The 'Details' tab for a P25 channel include a definition of the secondary control channels 
(primary is 'PCCH DOWNLINK', secondaries are 'SCCH DOWNLINK'). Perhaps that info could be used to 'autoswitch'

tmott1974, 2018-12-15:
Is there a chance this feature will get implemented? I travel frequently and use a private Icecast server to broadcast a few different talk groups so I can listen to my hometown wherever I am. The system where I live has 20 channels, three of which are marked as primary control channels and the remaining 17 are marked as alternate control channels. The problem is that the system seems to switch its control channel every 2 to 3 days and often to one of the alternate channels. This makes it difficult to track the system without frequent manual intervention.

DSheirer, 2018-12-17:
@tmott1974 I'm planning on resolving this issue for the 0.4.0 alpha 6 release.  I'm working on issue 434 at the moment, which has to be completed before I can implement a new channel/alias editor window. 

Resolving this issue requires a new channel frequency list editor and I'm planning to implement that as part of the new channel/alias editor window.   I've been holding off on this issue so that I don't have to do the work twice.  cheers, Denny

tmott1974, 2018-12-17:
Thank you Denny! That's great news.

DSheirer, 2019-01-06:
Feature is implemented with pull #475

closed 2019-01-06 by DSheirer