DSheirer/sdrtrunk #65: WaveWriter interrupted exception

DSheirer, 2016-04-06:
20160405 120705.526 [channel 1-136 pool-10-thread-1] ERROR r.wave.RealBufferWaveRecorder - Error starting real buffer recorder
java.nio.channels.ClosedByInterruptException: null
	at java.nio.channels.spi.AbstractInterruptibleChannel.end(AbstractInterruptibleChannel.java:202) ~[na:1.7.0_95]
	at sun.nio.ch.FileChannelImpl.write(FileChannelImpl.java:210) ~[na:1.7.0_95]
	at record.wave.WaveWriter.open(Unknown Source) ~[sdrtrunk.jar:na]
	at record.wave.WaveWriter.<init>(Unknown Source) ~[sdrtrunk.jar:na]
	at record.wave.WaveWriter.<init>(Unknown Source) ~[sdrtrunk.jar:na]
	at record.wave.RealBufferWaveRecorder.start(Unknown Source) [sdrtrunk.jar:na]
	at record.RecorderManager.receive(Unknown Source) [sdrtrunk.jar:na]
	at record.RecorderManager.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.audio.P25AudioModule.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.audio.P25AudioModule.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.Decoder.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.Decoder.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.Decoder.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageProcessor.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageProcessor.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer.dispatch(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer.access$300(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer$P25MessageAssembler.checkComplete(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer$P25MessageAssembler.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at dsp.psk.QPSKPolarSlicer.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.psk.QPSKPolarSlicer.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.psk.LSMDemodulator$GardnerSymbolTiming.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.psk.LSMDemodulator.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.gain.ComplexFeedForwardGainControl.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.complex.ComplexBufferToStreamConverter.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.complex.ComplexBufferToStreamConverter.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.fir.complex.ComplexFIRFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.fir.complex.ComplexFIRFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.halfband.complex.HalfBandFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.halfband.complex.HalfBandFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.fir.complex.ComplexFIRFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.fir.complex.ComplexFIRFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.complex.ComplexToComplexBufferAssembler.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$Output.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.decimator.ComplexDecimator.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$Stage.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$DecimatingStage.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.decimator.ComplexDecimator.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$TwoStage.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$DecimatingStage.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.decimator.ComplexDecimator.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$TwoStage.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$DecimatingStage.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate.receive(Unknown Source) [sdrtrunk.jar:na]
	at source.tuner.TunerChannelSource$DecimationProcessor.run(Unknown Source) [sdrtrunk.jar:na]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:471) [na:1.7.0_95]
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:304) [na:1.7.0_95]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:178) [na:1.7.0_95]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [na:1.7.0_95]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145) [na:1.7.0_95]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615) [na:1.7.0_95]
	at java.lang.Thread.run(Thread.java:745) [na:1.7.0_95]

Rylvir, 2016-04-18:
Haven't seen this pop up since I built from master after the April 06 Commit and have been running 24/7 since then.  Thanks

DSheirer, 2016-08-06:
RealBufferWaveRecorder extends Module and the abstract method start(ScheduledExecutorService) does not account for exceptions thrown on startup.  

Consider adding either a custom exception ModuleException to be thrown from this method so that modules like the recorder can throw a ModuleException wrapping an IOException and allow the module to be removed, or recreated.

closed 2017-01-07 by DSheirer