DSheirer/sdrtrunk #67: Tuner error

Tom-Neverwinter, 2016-04-17:
I had the program working just fine. Shut it down. Went to dinner. Powered back on and;
http://pastebin.com/HWGa9e5R

DSheirer, 2016-04-17:
It looks like another application or driver claimed the USB tuner after you shut down sdrtrunk the first time and so the tuner wasn't available to be claimed when you started sdrtrunk the second time.

Can you unplug the tuner, plug it back in, wait a few seconds for it to be recognized by the OS, and then start sdrtrunk, and see if the issue persists?

Tom-Neverwinter, 2016-04-17:
I removed all three sdr's and powered on, powered off again and plugged them in.
I tried removing them while the application was in focus then shutdown. powered on, powered off. plugged them back in.
powered on ran the application; ps all kill [pid of sdrtrunk] curiously two pids were there for the app. I killed both. 
sadly this did not solve it. 

"Thomas Edison failed more than 1,000 times when
trying to create the light bulb".  (The story is often told as 5,000
or 10,000 times depending on the version.)  When asked about it,
Edison allegedly said, "I have not failed 1,000 times.  I have
successfully discovered 1,000 ways to NOT make a light bulb."

closed 2017-01-28 by DSheirer