DSheirer/sdrtrunk #80: NBFM squelch & ctcss

imagesafari, 2016-10-16:
Looking for a way to record multiple analog conventional frequencies, ideally with squelch or CTCSS squelch. Could this be added?

DSheirer, 2016-10-19:
Ticket #36 is a feature request for NBFM squelch.
 
I'll rename this feature request to CSCSS squelching.

There are a number of tickets remaining for the 0.3.0 release that have to get done first, but then I can spend some time on these tickets.

nr0q, 2018-11-13:
Any updates on this enhancement? This would be great to allow SDRTrunk to also work to scan/stream conventional systems too!

denis-d, 2020-11-01:
Hope you will implement it soon. We have many mixed systems here.

smyers119, 2022-02-17:
Hi, I am interested in this as well.  Ideally ctcss/cdcss decoding would fall under a identifier in the alias section.  So it would give us a way to label the channel.  I would be interested in offering a bounty if that is an option

smyers119, 2022-02-19:
@DSheirer In case you didn't catch it, I added a bounty on this.  Thanks for your hard work.

DSheirer, 2023-06-10:
I think the latest power squelch is sufficient for squelching control.  However, to give more fine-grained control over how the decoded audio is handled, I'm going to add separate CDCSS and CTCSS decoders so that any audio generated from an NBFM squelch controlled channel can also be managed with Aliases where you can create aliases to control monitoring, recording and streaming of these calls.  

So, if you only want to hear audio from a repeater for specific squelching tones, you can control this with aliases, however the NBFM decoder will decode all audio that is transmitted on the repeater.

Closing this issue as not-implemented.

closed 2023-06-10 by DSheirer