DSheirer/sdrtrunk #83: Error: Could not find or load main class gui.SDRTrunk

xluccianox, 2017-01-04:
Hi after running the script ./scripts/run_sdrtrunk_windows.bat I get this error:
`Error: Could not find or load main class gui.SDRTrunk`

Im trying on Windows 8.1, here is my java -version:
```
C:\SDR\sdrtrunk-master\scripts>java -version
java version "1.8.0_91"
Java(TM) SE Runtime Environment (build 1.8.0_91-b15)
Java HotSpot(TM) 64-Bit Server VM (build 25.91-b15, mixed mode)
```

On Ubuntu 16.04 I have the same issue.

Thanks!

DSheirer, 2017-01-04:
This normally happens when you download the source files instead of the released version.  Try downloading the 0.2.0 release from the releases tab.  Please let me know if you have any more issues ... I'll close this ticket for now.

DSheirer, 2017-01-04:
Download link: https://github.com/DSheirer/sdrtrunk/releases/download/v0.2.0/sdrtrunk_0.2.0.tar.gz

closed 2017-01-04 by DSheirer

xluccianox, 2017-01-04:
Worked like a charm. Thanks for the quick reply and for sharing this tool.