DSheirer/sdrtrunk #92: Command line with no GUI

n2cr, 2017-01-07:
On an Odroid-C2 embedded ARM computer [http://www.hardkernel.com/main/products/prdt_info.php](url) the GUI takes too much CPU leaving it unable to decode the P25 Phase 1 system I stream to my local network.

Would it be possible to have a GUI-free command line option which would allow streaming?

Streaming working great on a virtual hosted on my local server!  Thanks so much for this feature.  Better than OP25 with GNURadio.

DSheirer, 2017-01-08:
This would be blocked on issues #34 and #53, both of which should improve the processing efficiency.  Both of these are the next big issues I'm going to tackle after I finish a few small bug issues.

ImagoTrigger, 2017-02-15:
playing around with JAVA_TOOL_OPTIONS: -Djava.awt.headless=true

i run multiple copies of sdrtrunk on the same server (using my dirty hacks) each instance uses 10-15% of my GPUs ;-)

chudgoo, 2017-02-15:
@ImagoTrigger  -  Is this running on an ODROID C2 or another aarch64 system?

ImagoTrigger, 2017-02-17:
nah, amd64 windhoes

n2cr, 2017-02-20:
Where does the JAVA_TOOL_OPTIONS go?

ImagoTrigger, 2017-02-28:
R:\scanner\out\production\sdrtrunk>java **-Djava.awt.headless=true** -XX:+UseG1GC -cp "SDRTrunk.jar;*;libs/*;config/*;images/*;imports/*;" gui.SDRTrunk

results in less resource usage. 0% GPU (tested on windows)

because of my poor java skills i just duped SDRTrunk class to make mMainGui a Jpanel instead of a JFrame when headless is defined.  of course you could reorganize to make it less fugly. also if headless dont do any JideTabbedPane stuff...  but this WORKS!!!! now my GPU does not heat up running SDRTrunk and now I can run it on a linux VPS....

`diff --git a/src/alias/AliasEditor.java b/src/alias/AliasEditor.java
index 706ef5e..2ee8276 100644
--- a/src/alias/AliasEditor.java
+++ b/src/alias/AliasEditor.java
@@ -93,15 +93,17 @@ public class AliasEditor extends Editor<Alias>
 		add( mAliasLabel, "align left" );
 		
 		add( new JSeparator( JSeparator.HORIZONTAL ), "span,growx,push" );
-		
-		JideTabbedPane tabs = new JideTabbedPane();
-		tabs.setFont( this.getFont() );
-    	tabs.setForeground( Color.BLACK );
-
-		tabs.addTab( "Alias", mAliasNameEditor );
-		tabs.addTab( "Audio / Identifier", mAliasIdentifierEditor );
-		tabs.addTab( "Action", mAliasActionEditor );
-		add( tabs, "span,grow" );
+
+		if (!java.awt.GraphicsEnvironment.isHeadless()) {
+			JideTabbedPane tabs = new JideTabbedPane();
+			tabs.setFont(this.getFont());
+			tabs.setForeground(Color.BLACK);
+
+			tabs.addTab("Alias", mAliasNameEditor);
+			tabs.addTab("Audio / Identifier", mAliasIdentifierEditor);
+			tabs.addTab("Action", mAliasActionEditor);
+			add(tabs, "span,grow");
+		}
 		
 		//Playlist management buttons
 		JButton btnSave = new JButton( "Save" );
diff --git a/src/controller/ControllerPanel.java b/src/controller/ControllerPanel.java
index 9b3f6cd..606168b 100644
--- a/src/controller/ControllerPanel.java
+++ b/src/controller/ControllerPanel.java
@@ -78,7 +78,7 @@ public class ControllerPanel extends JPanel
 
         mAudioPanel = new AudioPanel(iconManager, settingsManager, sourceManager, audioManager);
 
-        mChannelMetadataViewer = new ChannelMetadataViewer(channelProcessingManager, iconManager, settingsManager);
+        if (!java.awt.GraphicsEnvironment.isHeadless()) mChannelMetadataViewer = new ChannelMetadataViewer(channelProcessingManager, iconManager, settingsManager);
 
         mMapPanel = new MapPanel(mapService, iconManager, settingsManager);
 
@@ -98,21 +98,22 @@ public class ControllerPanel extends JPanel
         setLayout(new MigLayout("insets 0 0 0 0 ", "[grow,fill]", "[]0[grow,fill]0[]"));
 
         add(mAudioPanel, "wrap");
-
-        //Tabbed View - configuration, calls, messages, map
-        mTabbedPane = new JideTabbedPane();
-        mTabbedPane.setFont(this.getFont());
-        mTabbedPane.setForeground(Color.BLACK);
-        mTabbedPane.addTab("Now Playing", mChannelMetadataViewer);
-        mTabbedPane.addTab("Aliases", mAliasController);
-        mTabbedPane.addTab("Channels", mChannelController);
-        mTabbedPane.addTab("Map", mMapPanel);
-        mTabbedPane.addTab("Streaming", mBroadcastPanel);
-        mTabbedPane.addTab("Tuners", mTunerManagerPanel);
-
-        //Set preferred size to influence the split between these panels
-        mTabbedPane.setPreferredSize(new Dimension(880, 500));
-
-        add(mTabbedPane,"wrap");
[... 602 lines cut ...]
+            addActionListener(new ActionListener()
+            {
+                @Override
+                public void actionPerformed(ActionEvent e)
+                {
+                    toggleBroadcastStatusPanelVisibility();
+                    setSelected(mBroadcastStatusVisible);
+                }
+            });
+        }
+    }
+}
diff --git a/src/gui/control/JFrequencyControl.java b/src/gui/control/JFrequencyControl.java
index ca2053a..ed77b43 100644
--- a/src/gui/control/JFrequencyControl.java
+++ b/src/gui/control/JFrequencyControl.java
@@ -123,6 +123,7 @@ public class JFrequencyControl extends JPanel implements IFrequencyChangeProcess
         //Create image for cursor using empty array
         Image cursorImage = Toolkit.getDefaultToolkit().createImage(imageByte);
 
+        if (!java.awt.GraphicsEnvironment.isHeadless())
         mBlankCursor = Toolkit.getDefaultToolkit()
                 .createCustomCursor(cursorImage, new Point(0, 0), "cursor");
 
`

davidmckenzie, 2017-02-28:
That looks awesome, very keen to try it!

ImagoTrigger, 2017-03-01:
i was able to confirm this works on a very old server not running any kind of display manager.  I also commented out the default mixer stuff so java does not even try to load up any sound resources. instead of crapping out with the java.awt.HeadlessException it actually starts recording wavs!  this was on a early pentium 4 /w only 512mb ram.

chudgoo, 2017-09-25:
Hey @ImagoTrigger 
Can you provide some more detail about how to implement "crap-patch-no-gui"?  ;)

@DSheirer  Is there a way that a headless option could be made available using this method?
Use case scenario would be use GUI to set up your system (CCs, tuners, aliases, TGIDs, broadcastify, etc), and then use a separate bash script/sdrtrunk.jar that runs GUI-less from the terminal.  Maybe list call events in the terminal too?  Thoughts?

DSheirer, 2017-09-27:
A headless option is something we can do.

ImagoTrigger, 2017-09-28:
a soundless option would be cool also.

edbir1, 2018-04-23:
I am attempting to stream a P25 Phase1 system on a Raspberry Pi 3+ using 2 dongles.  It works, but I need to automagically start at boot, but I get the HeadlessException error which I think is complaining that I don't have a windows manager open.  I also noticed that if I close the gui window when I am running sdrtrunk from the desktop, the cpu usage drops to a usable level.  It is nice to have a trunking program which will run on a rpi!
1.  How can I run sdrtrunk at boot headless?

DSheirer, 2018-04-24:
There currently isn't a way to run sdrtrunk headless.

It would require a new class that performs all of the same startup operations that the SDRTrunk.java class performs, but without instantiating the gui pieces.

edbir1, 2018-04-24:
bummer.  
The ability to run headless and P25 Phase 2 are top on my Please list.
Thanks for writing the program and the fact that it will work on a rpi is awesome.

ImagoTrigger, 2018-10-25:
i made a nogui nosound patch right around v.3.  thus prevented jvm from
using gfx and snd libs completley.  basically i just copypastad .gui to
gui2 and commente out some jframe stuff....real fugly but it worked and it
ws awesome on a single core

> @obihann A CLI version and a smaller UI option both sound like great
features to add. You're more than welcome to help out with the coding if
you'd like to contribute.
>
> The CLI version might be fairly straight forward ... using the pieces
constructed in the SDRTrunk.java class, minus the gui pieces. It would have
to use the previously stored playlist (channels, aliases, auto-start)
initially. I'm completely revamping the way that aliasable values are
handled within the code, with an end goal of creating an interface to radio
reference to pull down networks and user/group info and also resolving a
number of issues with the current approach. I eventually intend to add
support for multiple playlists, so the CLI could be updated to accept an
optional playlist parameter down the road.
>
> A small UI that presents the minimum details would be neat. Are you
familiar with JavaFX? I want to transition the current UI to JavaFX, but
just haven't had the time yet.
https://ci4.googleusercontent.com/proxy/0Qp9VLoXO1ar3ONKjGAp_3GNWiOrnb_sIRfV-dXQjxlYEZLgE0yGxvAD5XwTnyUiOU7mXaKggKP8-OJKvBx8erJn5ts6K9MFf9DgbE8b-3PGuuHIq-cD0dn_cLUGEya4oLY25JDrej6XgZPeB5gm7Fk416LtAQ=s0-d-e1-ft#https://github.com/notifications/beacon/ADLfRFkvud6JzoELQEm19nhcfv3TyVzTks5uoXTwgaJpZM4Ldija.gif>

n2cr, 2020-05-16:
> 
> 
> A headless option is something we can do.

Is this still a possibility?  Is there a command line that would work with 0.4.0 or 0.5.0alpha and Java 13 that would work without the need for code modifications?
Using an Airspy and would like to make the startup into a service which doesn't require a GUI.

DSheirer, 2020-05-17:
It's still a possibility, but I haven't had any time to dedicte to it yet.

closed 2023-11-15 by DSheirer

n2cr, 2025-10-04:
You need to add “-Djava.awt.headless=true" to the Java arguments.  In the start script,  add:

set SDR_TRUNK_OPTS="-Djava.awt.headless=true"

right under the “set JAVA_EXE” line.

Works like a charm and reduces the load on small systems.