DSheirer/sdrtrunk #95: Feature: Audio Recording View

DSheirer, 2017-01-08:
Add a new tab to display audio recordings table with built-in audio player tool.

* Toggle showing only recordings generated during current application session or all recordings in the recording directory
* Allow user to delete recordings from this view.  
* Allow user to purge all recordings contained in the recording directory.

DSheirer, 2017-02-04:
From chudgoo:
Hello again~ This request has got to be the absolute farthest back burner type of thing... ;)

I think it would be really cool to have a "Call history" view similar to the control channel "events" view under "Now Playing", but instead of displaying live and recent events, it displays a searchable/filterable history of calls that you can click to play. Even better, select a range of calls or define in filters something like "show me all calls from TGID "04EA" between 10AM and 10:15AM on October 14th 2016", right click and export as either a collection of files in zip format or a single mp3 file of all of the selected calls merged into one. Thoughts?

(also, i realize that it's relatively easy to simply search through the recordings folder, but an in-app simple search / filter set would add a lot to the already impressive application, IMO)

thephantom1492, 2020-11-05:
It would be nice if there was also a web access for those recording, like trunkingrecorder have. SDRTrunk run on one computer, but I access the recording from another one...

DSheirer, 2023-11-15:
Closing.  Rescoped under issue #1723.

closed 2023-11-15 by DSheirer