DSheirer/sdrtrunk #96: Feature: ability to toggle FFT and spectral/waterfall display processing

DSheirer, 2017-01-08:
Add capability to suspend FFT processing and spectral/waterfall display.

Save user preference so that the setting persists across application sessions.

nr0q, 2018-01-13:
I'd like request bumping this up to med, I'm having some success running sdrtunk on an ITX computer with a Celeron-M processor, but the jmbe decoder crashes because I'm out of resources taken up by the waterfall/spectral display. I believe if I could turn it off, it would run better on lower end computers.

chudgoo, 2018-01-14:
Issue #92  is about a similar goal. Basically, running on very low end hardware.
Though people's intentions (including my own) were to run a streaming server, headless, without sound output. (easy enough... remove JMBE from the sdrtrunk dir)

PR #368 (2018-07-14): #96 Disable Spectral Display Option
Resolves #96

Adds capability to disable spectral display.

Adds right-click context
menu items to disable display and/or reenable display by selecting
another tuner.

Adds view menu items to disable spectral display or show a tuner.

Persists spectral display enabled state across application run sessions.
commit a8c649ef (2018-07-14): Resolves #96

Adds capability to disable spectral display.

Adds right-click context
menu items to disable display and/or reenable display by selecting
another tuner.

Adds view menu items to disable spectral display or show a tuner.

Persists spectral display enabled state across application run sessions.
closed 2018-07-14 by DSheirer