DSheirer/sdrtrunk #98: JMBE version check acting strange

chudgoo, 2017-01-09:
I'm running an instance of sdrtrunk cloned today (still 0.3.0 alpha, but after the streaming support was added).
I tried to use my old jmbe-0.3.2.jar jar file and got the following error any time an attempt to render audio is made.

15:51:38.964 WARN  m.d.p.a.P25AudioModule - JMBE library version 0.3.1 or higher is required - found: JMBE Audio Conversion Library v0.3.1

Then I tried the 0.3.3 version and am still getting the same error.
It seems to not be detecting the version of jmbe properly... or am I doing something super wrong?  ;) 
THanks~

DSheirer, 2017-01-09:
That's my fault ... I updated the code to check for version 0.3.3 or higher, but forgot to change the logging.  I'll have it fixed shortly.

DSheirer, 2017-01-09:
Should be fixed in code now.  JMBE version 0.3.3 changes the sample rate from 48 kHz down to 8 kHz.

closed 2017-01-09 by DSheirer