DSheirer/sdrtrunk #100: patch group streaming

grharrison583, 2017-01-10:
Hi, I have a talkgroup set up to stream to an icecast server, using the code updated yesterday. This talkgroup is normally part of a patch group, in the control channels event log the alias is logged correctly next to the patch group calls, but the majority of the time the voice channel doesn't display the alias, and its not streamed, but audio comes out of the local speakers correctly. the same streaming setup works correctly with a standalone talkgroup.

davidmckenzie, 2017-01-10:
Guessing this is P25?

I noticed this too during my testing - it seems to be that the audio processor is getting the talkgroup information from the data within the actual radio call, vs the event log which gets it from the data in the control channel. It's particularly evident in systems with poor reception.

Correct me if I'm wrong, but from memory the voice channel will send some information with every x frames of audio, including the talkgroup, radio IDs, etc. The data sent on the control channel side is only the talkgroup and channel number. If reception is poor, or the voice channel doesn't send call information regularly, then the voice channel won't "know" which talkgroup it's listening to.

I'm thinking this could be resolved by passing the event's talkgroup information to the voice channel when it is created, and putting some basic data validation on to prevent the voice channel replacing the event's talkgroup with something invalid (like 0000).

grharrison583, 2017-01-10:
Yes its a P25 motorola system. I think you are correct, except its only the patch group thats missing the information, all the standalone talkgroups are working correctly.

DSheirer, 2017-01-11:
You're correct ... the control channel indicates the talkgroup and channel number allocation and sdrtrunk passes that info to the allocated voice channel initially.  Once decoding of the voice channel starts, the FROM radio ID is picked up and added to the call event.  I allow the TO talkgroup and FROM radio ID to get updated based on decoding from the voice channel.  I probably need to clamp that down a bit and use something like a majority value as the final TO and FROM values.

For Patch groups ... when I receive a PATCH ADD message, I create a temporary Patch Alias that contains the patch talkgroup ID and one or more aliases matching the patched talkgroups, when they exist. Audio priority and streaming channels are rolled-up from each of the nested aliases so that they're treated as a single set of streaming channels for the patch group.  The audio priority is set to whatever the highest priority value is among the nested aliases.

However, if there is already an alias in the alias list that matches the Patch Talkgroup ID, I don't overwrite that alias.  In that situation, the patched audio priority and streaming chan is handled according to the patch alias.

I think I'm going to redo the Patch Alias management and add a placeholder for including the existing alias for the Patch group and treat that as the overall alias for the patch group. 

So, two separate issues:
1)Implement a better approach to assigning the TO/FROM talkgroup and radio ID values, and
2)Improve the patch group alias management.

Does this align with/cover the issues you're seeing, or is there another facet to this?

grharrison583, 2017-01-11:
yes I think that should fix it

DSheirer, 2017-01-11:
Issue number 1 is now #101

closed 2017-01-13 by DSheirer