DSheirer/sdrtrunk #104: Audio not streaming?

chudgoo, 2017-01-12:
So I entered by Broadcastify info, enabled and saved. 
The terminal window showed it as connected, the status under the streaming tab showed connected, but the audio was coming out locally (built-in output on a Macbook Pro).
When I connected to the broadcastify feed with both phone and a second laptop, I heard no audio. 

I tried the same thing using the IceCast 2.3 type with the same credentials and got the same results.

Have you seen this behavior before?  
Do I need to change the audio output settings? (right clicking in the area near the speaker icon)

davidmckenzie, 2017-01-12:
You need to add a "Broadcast Channel" item on the relevant alias entries. If you want to stream all audio from the system, use a wildcard alias. :)

DSheirer, 2017-01-13:
Were you able to get sound out of the stream?  When the broadcaster is connected and not sending audio, it sends silence MP3 frames to ensure the connection is maintained. 

Once you get your aliases setup as David indicated, you should start to see the numbers in the "Queued" and "Streamed" columns start to increase, and hear audio from the streaming client.

You can left click the green audio symbol in sdrtrunk to mute the local playback audio.  Click again to unmute.

chudgoo, 2017-01-13:
Oh! I didn't notice the new "Audio Broadcast Channel" option! Perfect. Working now. Thanks a lot guys! You're doing a hell of a job here Denny!  Close out issue... user error.  ;)

closed 2017-01-13 by DSheirer