DSheirer/sdrtrunk #117: Error submitting transfer [LIBUSB_ERROR_PIPE]

davidmckenzie, 2017-01-18:
Hi Denny!

I started using SDRTrunk a lot heavier yesterday due to fire activity. I added an extra couple of dongles and started tracking a P25 trunked system in addition to the two conventional P25 sites I'm monitoring. The errors in the below log start to pop up after it's been tracking for a couple of hours.

These errors only occur on the dongle that is following voice traffic. I've tested through multiple dongles and it doesn't matter which physical device it is, it's always the one that doesn't have the control channel or the other two conventional channels on it. Have tested with multiple bandwidths, gain, etc. 

The errors occur when the dongle is sent a command to change centre frequency. They start out benign, but eventually that dongle completely freezes in the FFT graph, and the only way to recover is to restart the app.

I'm guessing this is more a USB error than a bug in the code, though it may be related to #59 

One workaround I could see mitigating this is the ability to "lock" the centre frequency for individual dongles. I have full coverage of this system with two dongles, so there's really no need to change the centre frequency if it's set correctly.

As a side note - I'm also seeing errors of "Maximum recording time limit reached". What is the maximum time limit? Can we raise it? :)

Full log in comment to follow.

Thanks!
Dave

davidmckenzie, 2017-01-18:
`[18:38] davidm:~/sdrtrunk$ ./run_sdrtrunk_linux.sh
18:38:29.166 INFO  gui.SDRTrunk -
18:38:29.169 INFO  gui.SDRTrunk -
18:38:29.169 INFO  gui.SDRTrunk - *******************************************************************
18:38:29.169 INFO  gui.SDRTrunk - **** sdrtrunk: a trunked radio and digital decoding application ***
18:38:29.169 INFO  gui.SDRTrunk - ****  website: https://github.com/dsheirer/sdrtrunk             ***
18:38:29.170 INFO  gui.SDRTrunk - *******************************************************************
18:38:29.170 INFO  gui.SDRTrunk -
18:38:29.170 INFO  gui.SDRTrunk -
18:38:29.179 INFO  gui.SDRTrunk - Home path: /Users/davidm/SDRTrunk
18:38:29.180 INFO  p.SystemProperties - SystemProperties - loaded [/Users/davidm/SDRTrunk/SDRTrunk.properties]
18:38:29.180 INFO  p.SystemProperties - SystemProperties - application properties loaded [/Users/davidm/SDRTrunk/SDRTrunk.properties]
18:38:29.187 INFO  settings.SettingsManager - SettingsManager - loading settings file [/Users/davidm/SDRTrunk/settings/settings.xml]
18:38:29.415 INFO  source.mixer.MixerManager - loading system mixer devices
	[LOADED]     Input:  Default Audio Device CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: Default Audio Device CHANNELS: [MONO, STEREO]
	[LOADED]     Input:  Built-in Microphone CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: Built-in Output CHANNELS: [MONO, STEREO]

18:38:29.416 INFO  s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations
18:38:29.445 INFO  source.tuner.TunerManager - LibUSB API Version: 16777474
18:38:29.446 INFO  source.tuner.TunerManager - LibUSB Version: 1.0.18.10866
18:38:29.446 INFO  source.tuner.TunerManager - discovered [14] attached USB devices
18:38:31.189 INFO  source.tuner.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T 0010
18:38:32.911 INFO  source.tuner.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T 0101
18:38:34.630 INFO  source.tuner.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T 0100
18:38:34.630 INFO  source.tuner.TunerManager - usb device [05AC:8510] NOT LOADED: Unknown Device
18:38:34.631 INFO  source.tuner.TunerManager - usb device [05AC:8286] NOT LOADED: Unknown Device
18:38:34.631 INFO  source.tuner.TunerManager - usb device [05AC:0262] NOT LOADED: Unknown Device
18:38:34.631 INFO  source.tuner.TunerManager - usb device [0A5C:4500] NOT LOADED: Unknown Device
18:38:34.631 INFO  source.tuner.TunerManager - usb device [1A40:0101] NOT LOADED: Unknown Device
18:38:34.631 INFO  source.tuner.TunerManager - usb device [0424:2512] NOT LOADED: Unknown Device
18:38:34.632 INFO  source.tuner.TunerManager - usb device [8087:0024] NOT LOADED: Unknown Device
18:38:34.632 INFO  source.tuner.TunerManager - usb device [8087:0024] NOT LOADED: Unknown Device
18:38:34.632 INFO  source.tuner.TunerManager - usb device [05AC:8006] NOT LOADED: Unknown Device
18:38:34.632 INFO  source.tuner.TunerManager - usb device [05AC:8006] NOT LOADED: Unknown Device
18:38:34.632 INFO  source.tuner.TunerManager - usb device [05AC:8006] NOT LOADED: Unknown Device
18:38:36.230 INFO  icon.IconManager - loading icons file [/Users/davidm/SDRTrunk/settings/icons.xml]
18:38:36.380 INFO  playlist.PlaylistManager - loading version 2 playlist file [/Users/davidm/SDRTrunk/playlist/playlist_v2.xml]
18:38:36.707 INFO  m.d.p.a.P25AudioModule - JMBE audio conversion library [JMBE Audio Conversion Library v0.3.3] successfully loaded - P25 audio will be available
18:38:36.723 INFO  module.log.EventLogger - Creating log file:/Users/davidm/SDRTrunk/event_logs/20170118_183836_Canobolas_call_events.log
18:38:36.728 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Canobolas and Orana Zones] status: Connecting
18:38:36.728 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Fireground ] status: Connecting
18:38:36.749 INFO  module.log.EventLogger - Creating log file:/Users/davidm/SDRTrunk/event_logs/20170118_183836_W125_call_events.log
18:38:36.757 INFO  module.log.EventLogger - Creating log file:/Users/davidm/SDRTrunk/event_logs/20170118_183836_W056_call_events.log
18:38:36.763 INFO  gui.SDRTrunk - starting main application gui
18:38:37.714 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Canobolas and Orana Zones] status: Connected
18:38:37.714 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Fireground ] status: Connected
18:48:50.087 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
18:48:59.797 INFO  module.log.EventLogger - Creating log file:/Users/davidm/SDRTrunk/event_logs/20170118_184859_Canobolas_call_events.log
18:49:28.665 DEBUG a.broadcast.StreamManager - Maximum recording time limit reached - removing recorder: /Users/davidm/SDRTrunk/streaming/temporary_streaming_file_96_20170118_184840.594.mp3
18:49:28.665 DEBUG a.broadcast.StreamManager - Maximum recording time limit reached - removing recorder: /Users/davidm/SDRTrunk/streaming/temporary_streaming_file_98_20170118_184856.071.mp3
18:53:58.617 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
18:57:10.031 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
19:31:23.096 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
20:04:24.909 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
20:15:02.081 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
[... cut ...]
21:01:02.749 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
21:50:03.888 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
21:58:59.645 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
22:20:26.953 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
22:20:44.980 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]
23:57:36.716 ERROR s.t.r.RTL2832TunerController - Error submitting transfer [LIBUSB_ERROR_PIPE]`

DSheirer, 2017-01-19:
If this happens again with the USB errors and the FFT for the dongle locking up, could you please check memory usage and see how much memory it's using?  Does the program eventually crash?

For the recording time limits, I set an internal time limit of 30 seconds for streaming recordings.  I could let you change that, but the longer you allow the recorder to run, the longer  until the recording gets queued for streaming.

I'll put in another issue to expose the recorder time limit as a control for the stream configuration.

davidmckenzie, 2017-01-19:
Thanks mate! It occurred again about 5 hours ago, I only just noticed now. No crash, 4468M used by the java process, about 6G free on the box (10G used total). CPU is fairly stable at around 10% user, java process is showing about 80% of one core, load of 2. (This is on OSX, btw.)

Process started at 0440 local time, first USB transfer error was at 0447, last error at 1607, time now is 2122.

I did find some strange messages in the system.log:

```
Jan 19 11:01:32 stream-mbp spindump[560]: Saved wakeups_resource.diag report for java version ??? (???) to /Library/Logs/DiagnosticReports/java_2017-01-19-110132_stream-mbp.wakeups_resource.diag
Jan 19 11:15:59 stream-mbp spindump[560]: Saved wakeups_resource.diag report for java version ??? (???) to /Library/Logs/DiagnosticReports/java_2017-01-19-111559_stream-mbp.wakeups_resource.diag
Jan 19 11:23:50 stream-mbp spindump[560]: Saved wakeups_resource.diag report for java version ??? (???) to /Library/Logs/DiagnosticReports/java_2017-01-19-112350_stream-mbp.wakeups_resource.diag
Jan 19 11:30:12 stream-mbp spindump[560]: Saved wakeups_resource.diag report for java version ??? (???) to /Library/Logs/DiagnosticReports/java_2017-01-19-113012_stream-mbp.wakeups_resource.diag
Jan 19 11:37:57 stream-mbp spindump[560]: Saved wakeups_resource.diag report for java version ??? (???) to /Library/Logs/DiagnosticReports/java_2017-01-19-113757_stream-mbp.wakeups_resource.diag
```

I'll pop the most recent dump in the next comment.

Will leave the process up and running in the semi-hung state overnight, let me know if you need any further data while it's like this :)

davidmckenzie, 2017-01-19:
```
Date/Time:       2017-01-19 11:37:05 +1100
OS Version:      Mac OS X 10.12.1 (Build 16B2657)
Architecture:    x86_64
Report Version:  19

Command:         java
Path:            /Library/Java/JavaVirtualMachines/jdk1.8.0_111.jdk/Contents/Home/bin/java
Version:         ??? (???)
Parent:          bash [10958]
Responsible:     iTerm2 [236]
PID:             10959

Event:           wakeups
Wakeups:         45008 wakeups over the last 52 seconds (869 wakeups per second average), exceeding limit of 150 wakeups per second over 300 seconds
Duration:        51.82s
Steps:           67

Hardware model:  MacBookPro10,1
Active cpus:     8

Fan speed:       4936 rpm

Powerstats for:  java [10959]
UUID:            908B30B5-7ABB-3466-8743-E566104BBE9C
Start time:      2017-01-19 11:37:06 +1100
End time:        2017-01-19 11:37:56 +1100
Parent:          bash
Responsible:     iTerm2
Microstackshots: 61 samples (91%)
Primary state:   34 samples Frontmost App, User mode, Thread QoS Default
User Activity:   61 samples Idle, 0 samples Active
Power Source:    0 samples on Battery, 61 samples on AC
  14 ??? [0x1132230a4]
    14 ??? [0x11234c52a]
      14 Unsafe_Park + 126 (libjvm.dylib) [0x10e180a63]
        12 Parker::park(bool, long) + 512 (libjvm.dylib) [0x10e08656e]
          10 __psynch_cvwait + 10 (libsystem_kernel.dylib) [0x7fffac97ec8a]
            10 <Kernel mode>
          2  __gettimeofday + 17 (libsystem_kernel.dylib) [0x7fffac97e5ea]
            2  <Kernel mode>
        1  Parker::park(bool, long) + 563 (libjvm.dylib) [0x10e0865a1]
        1  Parker::park(bool, long) + 638 (libjvm.dylib) [0x10e0865ec]
          1  ThreadStateTransition::trans_and_fence(JavaThreadState, JavaThreadState) + 81 (libjvm.dylib) [0x10dd74e4d]
  7  ??? [0x113c1218c]
    7  ??? [0x11295526a]
      4  Java_sun_java2d_opengl_OGLRenderQueue_flushBuffer + 2886 (libawt_lwawt.dylib) [0x12e10382e]
        4  ??? (GeForceGLDriver + 3188755) [0x7fff92d09813]
          4  ??? (GeForceGLDriver + 3189143) [0x7fff92d09997]
            4  gpusSubmitDataBuffers + 156 (libGPUSupportMercury.dylib) [0x7fffa4f11466]
              2  IOAccelContextSubmitDataBuffersExt2 + 270 (IOAccelerator) [0x7fffa5f69b2d]
                2  IOConnectCallStructMethod + 56 (IOKit) [0x7fff994a29ac]
                  2  IOConnectCallMethod + 336 (IOKit) [0x7fff994a27e5]
                    2  io_connect_method + 375 (IOKit) [0x7fff9951cfae]
                      2  mach_msg_trap + 10 (libsystem_kernel.dylib) [0x7fffac97741a]
                        2  <Kernel mode>
              2  IOAccelContextSubmitDataBuffersExt2 + 587 (IOAccelerator) [0x7fffa5f69c6a]
                2  tiny_free_no_lock + 1 (libsystem_malloc.dylib) [0x7fffac9e9367]
      2  Java_sun_java2d_opengl_OGLRenderQueue_flushBuffer + 910 (libawt_lwawt.dylib) [0x12e103076]
        2  OGLBlitLoops_IsoBlit + 905 (libawt_lwawt.dylib) [0x12e0fd010]
          2  glBegin_Exec + 96 (GLEngine) [0x7fff9c38bbdc]
            2  gleDoDrawDispatchCore + 520 (GLEngine) [0x7fff9c47d0bd]
              2  gldUpdateDispatch + 1101 (GeForceGLDriver) [0x7fff92d09319]
                2  ??? (GeForceGLDriver + 2032205) [0x7fff92bef24d]
                  2  ??? (GeForceGLDriver + 6403543) [0x7fff9301a5d7]
                    2  ??? (GeForceGLDriver + 6403302) [0x7fff9301a4e6]
                      2  ??? (GeForceGLDriver + 6402414) [0x7fff9301a16e]
      1  Java_sun_java2d_opengl_OGLRenderQueue_flushBuffer + 2862 (libawt_lwawt.dylib) [0x12e103816]
        1  OGLRenderQueue_CheckPreviousOp + 127 (libawt_lwawt.dylib) [0x12e103aad]
          1  OGLVertexCache_DisableMaskCache + 17 (libawt_lwawt.dylib) [0x12e107611]
[... 101 lines cut ...]
    1  ??? [0x112c72e7f]
      1  libusb_submit_transfer + 311 (libusb4java.dylib) [0x1336800a7]
        1  __psynch_mutexwait + 10 (libsystem_kernel.dylib) [0x7fffac97ecba]
          1  <Kernel mode>
  1  ??? [0x1122b9678]

  Binary Images:
         0x10c9ae000 -        0x10c9bffef  java <908B30B5-7ABB-3466-8743-E566104BBE9C> /Library/Java/JavaVirtualMachines/jdk1.8.0_111.jdk/Contents/Home/bin/java
         0x10dc00000 -        0x10e40bfff  libjvm.dylib (1) <FC87AA86-AA04-38D6-9EA7-BFAD6748E29B> /Library/Java/JavaVirtualMachines/jdk1.8.0_111.jdk/Contents/Home/jre/lib/server/libjvm.dylib
         0x12df54000 -        0x12dfbdfff  libawt.dylib (1) <D2B56651-0C7C-3ADF-8F74-A95F040BB301> /Library/Java/JavaVirtualMachines/jdk1.8.0_111.jdk/Contents/Home/jre/lib/libawt.dylib
         0x12e0c8000 -        0x12e138fff  libawt_lwawt.dylib (1) <A7B6FF26-BC4A-35DF-B3DD-2EFA9D14B0FF> /Library/Java/JavaVirtualMachines/jdk1.8.0_111.jdk/Contents/Home/jre/lib/libawt_lwawt.dylib
         0x130064000 -        0x130088ff7  libdcpr.dylib (1) <2122AAE9-CBF8-36A2-8F44-04777D4A4CF9> /Library/Java/JavaVirtualMachines/jdk1.8.0_111.jdk/Contents/Home/jre/lib/libdcpr.dylib
         0x133675000 -        0x13368cff7  libusb4java.dylib (0) <FDB7A47A-4289-362B-9639-C1D5F4D0662D> /private/var/folders/*/libusb4java.dylib
      0x7fff929ff000 -     0x7fff9324bfff  com.apple.GeForceGLDriver 10.12.68 (10.1.2) <67020676-5F42-3BD7-AE17-610E00DF38B4> /System/Library/Extensions/GeForceGLDriver.bundle/Contents/MacOS/GeForceGLDriver
      0x7fff98c0a000 -     0x7fff98c3ffff  com.apple.CoreVideo 1.8 (234.0) <48C31E93-87C2-31F4-97E7-9E54C1EA8E7D> /System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo
[... cut ...]
      0x7fffa5f67000 -     0x7fffa5f6effb  com.apple.IOAccelerator 288.15 (288.15) <2197157C-B721-362B-BF5B-E2A844A4E345> /System/Library/PrivateFrameworks/IOAccelerator.framework/Versions/A/IOAccelerator
      0x7fffac965000 -     0x7fffac987ff7  libsystem_kernel.dylib (3789.21.4) <651F249F-D797-3FBD-A66F-66DFB48FCF38> /usr/lib/system/libsystem_kernel.dylib
      0x7fffac9d0000 -     0x7fffac9eeff7  libsystem_malloc.dylib (116) <3DD17B88-B7A4-38B9-9E95-AB88E1C3B647> /usr/lib/system/libsystem_malloc.dylib
      0x7fffaca64000 -     0x7fffaca6eff7  libsystem_pthread.dylib (218.20.1) <F35891AC-CA82-3F97-9BE6-15EC3B4C36F6> /usr/lib/system/libsystem_pthread.dylib
```

davidmckenzie, 2017-01-19:
Just another observation - glanced over and saw the stalled dongle had a channel active on it. It registered a call event on a channel within that dongle's bandwidth, however it wasn't producing any audio. The channel indicator would flash on and off on the screen about twice a second (as if the channel was getting created and destroyed repeatedly).

DSheirer, 2017-01-20:
I was seeing a possibly related issue today while streaming 2 broadcastify channels that I was over-feeding with too many calls, causing the streaming queue to build up.  Several times the application thread locked and caused out of memory.  I think it's caused by the streaming queue processor.  I'll put a new issue, fix it, and see if that resolves what you're seeing.

DSheirer, 2017-01-24:
Related to issue #126

DSheirer, 2017-01-28:
Related issue #126 is now resolved/closed with merge #142.  Closing this issue.  Please advise if you see any more related to this.

davidmckenzie, 2017-01-28:
Thanks @DSheirer!

Unfortunately looks like an increase in errors since the latest commit, though at least this time it's identifying the responsible dongle.

The dongle is the one that's dedicated to following the trunked voice channels (whereas one other is on a conventional P25 channel, and the other is on the control channel).

I notice that it's only grabbing 256M RAM out of a max of 4G - wonder if it would be worth tuning the min/max heap size in run_sdrtrunk_linux.sh?

```
08:00:52.738 INFO  gui.SDRTrunk -
08:00:52.741 INFO  gui.SDRTrunk - *******************************************************************
08:00:52.742 INFO  gui.SDRTrunk - **** sdrtrunk: a trunked radio and digital decoding application ***
08:00:52.742 INFO  gui.SDRTrunk - ****  website: https://github.com/dsheirer/sdrtrunk             ***
08:00:52.742 INFO  gui.SDRTrunk - *******************************************************************
08:00:52.742 INFO  gui.SDRTrunk -
08:00:52.742 INFO  gui.SDRTrunk - Host CPU Cores: 8
08:00:52.742 INFO  gui.SDRTrunk - Host Memory Total: 268435456
08:00:52.742 INFO  gui.SDRTrunk - Host Memory Max: 4294967296
08:00:52.742 INFO  gui.SDRTrunk - Host Memory Free: 252201408
08:00:52.775 INFO  util.ThreadPool - Application thread pool created with [8] threads
08:00:52.775 INFO  gui.SDRTrunk - Home path: /Users/davidm/SDRTrunk
08:00:52.777 INFO  p.SystemProperties - SystemProperties - loaded [/Users/davidm/SDRTrunk/SDRTrunk.properties]
08:00:52.777 INFO  p.SystemProperties - SystemProperties - application properties loaded [/Users/davidm/SDRTrunk/SDRTrunk.properties]
08:00:52.790 INFO  settings.SettingsManager - SettingsManager - loading settings file [/Users/davidm/SDRTrunk/settings/settings.xml]
08:00:53.191 INFO  source.mixer.MixerManager - loading system mixer devices
	[LOADED]     Input:  Default Audio Device CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: Default Audio Device CHANNELS: [MONO, STEREO]
	[LOADED]     Input:  Built-in Microphone CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: Built-in Output CHANNELS: [MONO, STEREO]

08:00:53.192 INFO  s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations
08:00:53.239 INFO  source.tuner.TunerManager - LibUSB API Version: 16777474
08:00:53.240 INFO  source.tuner.TunerManager - LibUSB Version: 1.0.18.10866
08:00:53.240 INFO  source.tuner.TunerManager - discovered [15] attached USB devices
08:00:54.990 INFO  source.tuner.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T 0011
08:00:56.686 INFO  source.tuner.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T 0010
08:00:58.393 INFO  source.tuner.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T 0100
08:00:58.394 INFO  source.tuner.TunerManager - usb device [05AC:8510] NOT LOADED: Unknown Device
08:00:58.394 INFO  source.tuner.TunerManager - usb device [05AC:8286] NOT LOADED: Unknown Device
08:00:58.394 INFO  source.tuner.TunerManager - usb device [05AC:0262] NOT LOADED: Unknown Device
08:00:58.394 INFO  source.tuner.TunerManager - usb device [0A5C:4500] NOT LOADED: Unknown Device
08:00:58.395 INFO  source.tuner.TunerManager - usb device [1A40:0101] NOT LOADED: Unknown Device
08:00:58.395 INFO  source.tuner.TunerManager - usb device [0424:2512] NOT LOADED: Unknown Device
08:00:58.395 INFO  source.tuner.TunerManager - usb device [8087:0024] NOT LOADED: Unknown Device
08:00:58.395 INFO  source.tuner.TunerManager - usb device [8087:0024] NOT LOADED: Unknown Device
08:00:58.395 INFO  source.tuner.TunerManager - usb device [05AC:8006] NOT LOADED: Unknown Device
08:00:58.396 INFO  source.tuner.TunerManager - usb device [05AC:8006] NOT LOADED: Unknown Device
08:00:58.396 INFO  source.tuner.TunerManager - usb device [05AC:8006] NOT LOADED: Unknown Device
08:01:00.911 INFO  icon.IconManager - loading icons file [/Users/davidm/SDRTrunk/settings/icons.xml]
08:01:01.103 INFO  playlist.PlaylistManager - Attempting to load version 2 playlist file [/Users/davidm/SDRTrunk/playlist/playlist_v2.xml]
08:01:01.448 INFO  m.d.p.a.P25AudioModule - JMBE audio conversion library [JMBE Audio Conversion Library v0.3.3] successfully loaded - P25 audio will be available
08:01:01.467 INFO  module.log.EventLogger - Creating log file:/Users/davidm/SDRTrunk/event_logs/20170129_080101_Canobolas_call_events.log
08:01:01.491 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connecting
08:01:01.502 INFO  gui.SDRTrunk - starting main application gui
08:01:04.199 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connected

08:14:02.273 ERROR s.t.u.USBTransferProcessor - R820T 0100 - error submitting transfer [LIBUSB_ERROR_PIPE]
08:14:02.275 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:14:02.276 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:14:02.276 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:14:02.276 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:14:02.276 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:14:02.276 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:14:02.276 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
[... cut ...]
08:46:45.623 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:46:45.623 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:46:45.624 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:46:45.624 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
08:46:45.624 ERROR s.t.u.USBTransferProcessor - R820T 0100- error submitting transfer [LIBUSB_ERROR_PIPE]
```

DSheirer, 2017-01-28:
I just merged pull request #155 that hopefully resolves this issue for you.  Can you please test and let me know if it resolves the issue for you?

davidmckenzie, 2017-01-28:
Cheers mate, compiling now :)

Side note - I noticed the errors /only/ occur when the device is not the active dongle in the waterfall. If that dongle was visible in the waterfall, then no errors.

davidmckenzie, 2017-01-29:
Errors are much less frequent again on the latest commit, but still present. As with the last one - they only occur if the tuner in question is not visible on the waterfall, so I guess it's something to do with the way changes to centre frequency redraw the waterfall?

```
10:25:17.371 ERROR s.t.u.USBTransferProcessor - R820T 0010 - error submitting transfer [LIBUSB_ERROR_PIPE]
10:32:24.863 ERROR s.t.u.USBTransferProcessor - R820T 0010 - error submitting transfer [LIBUSB_ERROR_PIPE]
10:37:55.475 ERROR s.t.u.USBTransferProcessor - R820T 0010 - error submitting transfer [LIBUSB_ERROR_PIPE]
10:42:59.671 ERROR s.t.u.USBTransferProcessor - R820T 0010 - error submitting transfer [LIBUSB_ERROR_PIPE]
10:55:34.696 ERROR s.t.u.USBTransferProcessor - R820T 0010 - error submitting transfer [LIBUSB_ERROR_PIPE]
```

I've now got this tuner showing in the waterfall, so will monitor for the next day or so and see if any errors pop up while it's visible :)

DSheirer, 2017-01-29:
Ok, I just merged one more pull request to add a pipe error handler in the transfer processing section.  Can you test this one out to see if it clears the final set of errors?

davidmckenzie, 2017-01-29:
Thanks again! Looking good so far, will continue to monitor for a couple of hours before closing this.

davidmckenzie, 2017-01-29:
No errors so far so will close this off :)

closed 2017-01-29 by davidmckenzie