DSheirer/sdrtrunk #119: Add capability to blacklist Tuners by serial number

DSheirer, 2017-01-20:
Add capability for user to select a tuner and designate it as 'blacklisted', meaning that the tuner should not be used by the application.

Since this can only be done if/when the application is aware of the tuner, add the capability to the 'Tuners' pane and indicate to the user that they must restart the application for the setting to take effect.

On startup, the TunerManager should verify each tuner against the blacklisting and release the USB device when a tuner is blacklisted.

DSheirer, 2018-03-20:
Also add the capability to manually 'unload' a tuner that has be opened on application startup.

ImagoTrigger, 2020-03-01:
we could piggy back #172 to this if a user sets the tuner range as 0mhz-0mhz or something along that lines (then unload/or don't load that tuner).  i just personally realized how having #172 features would really improve my scanner quality

DSheirer, 2022-04-24:
Added in version 0.5.0 beta 3.

closed 2022-04-24 by DSheirer