DSheirer/sdrtrunk #132: Broadcastify error

chudgoo, 2017-01-23:
I've had an issue the past couple of days where the connection to Broadcastify drops. 
Not exactly sure what the cause is... The network was up the entire time (I've also got a Moto Type II system feeding out of here and it never dropped).
Here's the text from the launch window.

21:55:42.104 ERROR a.b.i.IcecastTCPAudioBroadcaster - Broadcast error
java.io.IOException: Operation timed out
	at sun.nio.ch.FileDispatcherImpl.read0(Native Method) ~[na:1.8.0_102]
	at sun.nio.ch.SocketDispatcher.read(SocketDispatcher.java:39) ~[na:1.8.0_102]
	at sun.nio.ch.IOUtil.readIntoNativeBuffer(IOUtil.java:223) ~[na:1.8.0_102]
	at sun.nio.ch.IOUtil.read(IOUtil.java:197) ~[na:1.8.0_102]
	at sun.nio.ch.SocketChannelImpl.read(SocketChannelImpl.java:380) ~[na:1.8.0_102]
	at org.apache.mina.transport.socket.nio.NioProcessor.read(NioProcessor.java:317) ~[mina-core-2.0.16.jar:na]
	at org.apache.mina.transport.socket.nio.NioProcessor.read(NioProcessor.java:45) ~[mina-core-2.0.16.jar:na]
	at org.apache.mina.core.polling.AbstractPollingIoProcessor.read(AbstractPollingIoProcessor.java:683) [mina-core-2.0.16.jar:na]
	at org.apache.mina.core.polling.AbstractPollingIoProcessor.process(AbstractPollingIoProcessor.java:659) [mina-core-2.0.16.jar:na]
	at org.apache.mina.core.polling.AbstractPollingIoProcessor.process(AbstractPollingIoProcessor.java:648) [mina-core-2.0.16.jar:na]
	at org.apache.mina.core.polling.AbstractPollingIoProcessor.access$600(AbstractPollingIoProcessor.java:68) [mina-core-2.0.16.jar:na]
	at org.apache.mina.core.polling.AbstractPollingIoProcessor$Processor.run(AbstractPollingIoProcessor.java:1120) [mina-core-2.0.16.jar:na]
	at org.apache.mina.util.NamePreservingRunnable.run(NamePreservingRunnable.java:64) [mina-core-2.0.16.jar:na]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142) [na:1.8.0_102]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617) [na:1.8.0_102]
	at java.lang.Thread.run(Thread.java:745) [na:1.8.0_102]
21:55:42.107 INFO  a.b.AudioBroadcaster - [BLAH] status: Error

davidmckenzie, 2017-01-23:
That looks like a TCP timeout - is your other system streaming from SDRTrunk as well? Is it hitting the same hostname? (i.e. audiox.broadcastify.com)

I've noticed the same error on my system but that was during actual internet drop-outs. I noted that it was not very resilient to timeout errors only. If the stream consumer sends a connection reset, it attempts reconnection a few times before giving up.

I think if the netcode was able to catch the timeouts and treat them the same way it treats connection resets, then it should solve this.

chudgoo, 2017-01-23:
The other system (believe it or not) is an old Radioshack Pro-95 connected to a $5 android TracFone running the BroadcastMe icecast streaming app.  It's a very different setup... but it is connecting to the same audiox.broadcastify.com server.

davidmckenzie, 2017-01-24:
Just further to this - I experienced a different network error last night. There was no timeout or conn reset shown in the logs, just a "disconnected":

```
19:30:00.880 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Disconnected
19:30:01.485 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connecting
19:30:05.642 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connected
19:31:00.170 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Disconnected
19:31:00.484 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connecting
```
It then did nothing until I manually reconnected:
```
21:06:11.421 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connecting
21:06:12.007 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connected
```

It seems my terrible internet connection provides some exotic disconnect conditions, so if you need someone to test anything in this space, I'd be happy to oblige :)

FWIW, this is what a connection reset looks like:

```
13:27:55.355 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connected
13:38:26.320 INFO  a.b.i.IcecastTCPAudioBroadcaster - Streaming connection reset by remote server - reestablishing connection
13:38:26.320 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Disconnected
13:38:26.540 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connecting
13:38:36.607 ERROR a.b.i.IcecastTCPAudioBroadcaster - Unrecognized server response:HTTP/1.1 302 Moved Temporarily
13:38:36.607 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Error
```

(Disregard the 302 - that was due to one of those "captive portal" router things.)

DSheirer, 2017-01-25:
I just merged a pull request to add error handler for socket timeouts.  When detected, it will reset the connection (disconnect/connect).  Could you please pull down the latest and test and let me know if any issues with connectivity still remain?  Thanks very much for reporting/testing!

davidmckenzie, 2017-01-25:
Thanks mate! Running on the new code now, will see how it goes.

Just before I compiled, I got a different network error during a short period of packet loss. I know it's not relevant to this particular error, but figured worth sharing :)

```
11:37:52.167 INFO  a.b.i.IcecastTCPAudioBroadcaster - Streaming connection reset by remote server - reestablishing connection
11:37:52.167 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Disconnected
11:37:52.416 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connecting
11:37:53.005 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Mount Point In Use
11:37:53.005 ERROR a.b.i.IcecastTCPAudioBroadcaster - Unrecognized server response:Content-Type: text/html
11:37:53.005 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Error
```

DSheirer, 2017-01-25:
The 'Mount Point In Use' error means the remote server thinks that someone is already streaming.  It should set the state to error.  Usually this will happen if you disconnect and try to reconnect too quickly before the remote server has a chance to discover that you've disconnected.

davidmckenzie, 2017-01-27:
Got a timeout last night, looks good here :)

Thanks Denny!

```
02:01:03.631 INFO  a.b.i.IcecastTCPAudioBroadcaster - Streaming connection timed out - resetting connection
02:01:03.636 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Disconnected
02:01:04.634 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connecting
02:01:05.129 INFO  a.b.AudioBroadcaster - [NSW Rural Fire Service - Central West] status: Connected
```

DSheirer, 2017-01-28:
Closing issue.

closed 2017-01-28 by DSheirer