DSheirer/sdrtrunk #133: Feature request - Audio compressor, limiter, normalization options

chudgoo, 2017-01-23:
On my local P25 system we've got a cast of characters that either eat their microphones and scream or whisper like they're afraid of being heard. 

I was playing with piping sdrtrunk audio into some stand alone audio VSTs for compression/limiting and it worked fairly well. Then I found this java based VST wrapper that could potentially allow the use of other open source VST audio effects.

I think it would be awesome if there was an 'audio post processing' tab to allow for some options on what the output audio sounds like.  Might be a colossal pain to implement, I have no idea what it would entail on your end, but think it would help the already great sounding decoded audio be more consistent.
Thoughts?

http://jvstwrapper.sourceforge.net/

DSheirer, 2017-01-24:
That might be usable.  Looks like the code hasn't been updated since 2009, but if it's a solid product, maybe we can make use of it.

chudgoo, 2017-01-24:
It seems there are quite a few things out there that do java VST hosting...

(5 years since last update)
"JVST provides an interface to VST plugins (effects and instruments) for use in Java. It allows the developer to interact with a plugin, process samples and use any built in GUIs. "   - That last part is pretty interesting for seamless integration with the sdrtrunk GUI.
http://jvst.sourceforge.net/

"A Java-based VST Host. VST instruments and effects can be loaded and manipulated by this library in Linux, OS X, and Windows."
https://github.com/mhroth/jvsthost

This one is standalone but could possibly work.
http://www.hotto.de/software/audiocompressorlimiter.html