DSheirer/sdrtrunk #137: OutOfMemoryError exception thrown when audio is muted

grumble1965, 2017-01-25:
When I leave the audio muted for a period of time, the application eventually fails with an out of memory exception.  This doesn't happen if I leave the audio playing.  I'd guess there's a code path when muted where a sample isn't being collected.

Running on windows 10 with an Airspy.

> Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "D3D Screen Updater"
> Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk pool-1-thread-2"
> Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk pool-1-thread-1"
> Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "AWT-EventQueue-0"
> Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "Airspy Buffer Processor"

DSheirer, 2017-01-26:
This issue is probably the same as #126 and it is likely caused by the airspy tuner filling an unbounded buffer with samples (at 10 MHz) while java stops the consumer thread during an extended garbage collection phase.  This is the next bug in the hopper to fix and I hope to have it resolved by this weekend.

The audio output(s) continuously write samples to the mixer and rely on the mixer itselft to either mute the channel or not.  So, there shouldn't be any path switching that depends on the mute setting.

DSheirer, 2017-01-28:
I just committed the fix for #126.  I'm going to close this issue.  Please let me know if you are still seeing issues with this out of memory error.

closed 2017-01-28 by DSheirer

grumble1965, 2017-01-30:
Just built the latest commits - works like a champ!  Thanks for the help!