DSheirer/sdrtrunk #153: Spectral view rendering artifacts from cursor appended frequency text

DSheirer, 2017-01-28:
If you access the menus above the spectral display, the appended frequency label leaves artifacts on the panel just below the menu items.

thewraith2008, 2017-02-25:
Hi Denny,

I've been investigating this problem today and narrowed it down to the SpectralDisplayPanel class.
Specifically the mouse overlay that is used in 'mOverlayPanel' for the spectral display.

It seems that when selecting a menu item the mouse re-enters the mOverlayPanel and leaves this frequency value artifact in the JMenuBar. The how and why is beyond me.

If I add the following to line 580, before the IF statement in the 'mouseEntered' method in 'SpectralDisplayPanel' class, it cures this problem.

> mOverlayPanel.dispose();

Hope this is a suitable solution for this issue.

DSheirer, 2017-02-26:
I had to add code to measure the height of the frequency label and then check the vertical location of the cursor so that that frequency label wasn't painted anytime the cursor vertical location would cause the frequency label to be rendered outside of the spectral display.

closed 2017-02-26 by DSheirer

thewraith2008, 2017-02-26:
Thanks for spending time on this. I realized earlier when I started getting errors with the tuner that using '.dispose' was not right. I was just looking into someway to refresh the JMenuBar when I got this notification of this merge.

All looks good now. Cheers.

PR #209 (2017-02-26): Resolves issue with spectral display rendering artifacts
Resolves #153 issue with spectral display rendering artifacts from cursor frequency readout when accessing menu items directly above the spectral display.