DSheirer/sdrtrunk #165: Second decoding channel stops when decoding 2 or more channels

DSheirer, 2017-01-30:
I'm decoding a P25 control channel on a 15-channel system and also decoding an MPT1327 control channel on a 5-channel system at the same time.  At some point after a few hours, the P25 control channel simply stops decoding.

There is logging that indicates that this may be related to overflow condition in the OverflowableTransferQueue.  Possibly a threading issue.

whelgeson, 2017-02-06:
I'm seeing this issue as well on the current master branch while decoding a single P25 control channel.  

Console prints: "Channel [16954673440] buffer overflow cleared - resuming normal processing" 
Then processing stops and I have to disable and re-enable the channel to get it to resume decoding.

DSheirer, 2017-02-07:
This seems to be primarily affecting P25 channels ... initial thought is that the P25 decoder's fine sync control may be retuning the signal out of bounds during overflow such that the decoder is unable to pull it back.  Going to experiment with resetting fine sync control in conjunction with overflow reset.

DSheirer, 2017-02-08:
Once the lockup on the P25 channel happens, the TunerChannelSource is still receiving samples from the tuner, but the DecimationProcessor is no longer running (no samples to the P25 decoder).  I'm not sure if the decimation processor stops running first, causing the buffer to back up, or if the buffer backs up and then for some reason the decimation processor stops.  Still working on this.

DSheirer, 2017-02-10:
Found the issue to be with patch group management.  Stale patch groups were not being cleaned up and under certain situations were creating circular dependencies when a patch group was added to itself as a patched group, causing stack overflow when attempting to access patch group metadata like call priority or broadcast channels.  This causes the buffer consumer thread to die and the incoming sample buffer to overflow.  The channel never recovers and reflects an IDLE state.

closed 2017-02-10 by DSheirer

sdrtrunkuser, 2017-03-26:
Had this with 0.3.0b9 today. (2 cheap Amazon tuners connected to a USB3 hub.)
Dispatcher announced emergency traffic.
Channel state changed from "CONTROL" to "OVERFLOW."
Traffic channel carrying the audio stuck in a "TEARDOWN" state and the control channel never left OVERFLOW state.

I am seeing buffer overflow cleared messages fairly regularly in the console.
    DEBUG s.t.u.USBTransferPRocessor - R820T 11111111 - buffer overflow cleared - resuming normal processing

Only happened once, but if I can enable additional logging for next time, please let me know.

davidmckenzie, 2017-03-26:
USB2 hubs are generally better for RTL-SDRs - also make sure you're using an externally powered hub. I've seen weird issues when attempting to use USB3 hubs, and you don't get any additional USB bandwidth by using USB2 devices on a 3 hub.

sdrtrunkuser, 2017-03-26:
Thanks! I'll try to dig up a USB2 hub to test with or find some usb extensions so I can plug them in directly. Powering this [Amazon Basics USB3 hub](https://www.amazon.com/AmazonBasics-Port-2-5A-power-adapter/dp/B00DQFGH80) did not appear to help. Would the sample rate setting on the tuners affect this as well? (or traffic pool size?) After watching it a bit more, it does appear tuner2 covering a channel allocated a frequency a couple mhz above the control channel on tuner1 is somehow related to the messsages.

davidmckenzie, 2017-03-26:
Yeah you'll wanna experiment with sample rate until you find the right one - for me, I can only choose .96 or 2.4MHz, anything else results in strange behaviour.

Some other tips I've found help with stability - have the scope/waterfall showing your 'dynamic' dongle, i.e. the one that's not monitoring the control channel, and will move around a lot; keep the fft width low where you can (I use 2048), and keep the frame rate around 20/25. Those last two settings can be accessed by right clicking the waterfall.

Hope that helps!

DSheirer, 2017-03-31:
@sdrtrunkuser are you still having issues with the overflow state when running 2 dongles?

sdrtrunkuser, 2017-03-31:
Yes. I will try to dig up my usb2 powered hub to try and will also plug the tuners in directly to compare.

Currently I have 3 tuners connected:
Tuner1 - via Amazon USB3 hub (serial 111)
Tuner2 - via Amazon USB3 hub (serial 222)
Tuner3 - Direct USB (serial 333)

SDRTrunk seems to always grab the non-hub tuner as "primary" on startup. With all three connected, I don't actually see the 3rd tuner ever need to tune a channel so far. (Evidently two tuners is all that is really need to cover the local P25 traffic.)

Even though SDRTrunk is only showing channels active on Tuner3(primary) and Tuner2, I still see the USB overflow errors regularly.

sdrtrunkuser, 2017-03-31:
With one device directly connected in the back, and one in the front on a USB connected to a USB header on the motherboard, I'm still getting these messages. Maybe I've got a driver issue? [Here's my USB setup from device manager](http://imgur.com/a/WqiSs) Next I'll try a USB2 hub.

DSheirer, 2017-03-31:
@sdrtrunkuser what processor and memory does your computer have and how many channels do you have enabled?  The overflow messages generally mean that your computer is not keeping up with the processing workload.

sdrtrunkuser, 2017-03-31:
Processor: I7-2600k
Memory: 16GB
Channels enabled: 1
Traffic Pool: 8
Decoder: P25 Phase 1
Modulation: Simulcast (LSM)

I see the messages even with CPU not spiking above ~15% and memory never exceeding 62%.

crushkittykitty, 2017-05-03:
07:20:52.578 WARN  audio.output.AudioOutput - Couldn't obtain MASTER GAIN control for stereo line [PulseAudio Mixer | LEFT]
07:20:52.578 WARN  audio.output.AudioOutput - Couldn't obtain MUTE control for stereo line [PulseAudio Mixer | LEFT]
07:20:52.580 WARN  audio.output.AudioOutput - Couldn't obtain MASTER GAIN control for stereo line [PulseAudio Mixer | RIGHT]
07:20:52.581 WARN  audio.output.AudioOutput - Couldn't obtain MUTE control for stereo line [PulseAudio Mixer | RIGHT]
07:20:54.709 INFO  icon.IconManager - loading icons file [/home/ps4/SDRTrunk/settings/icons.xml]
07:20:55.138 INFO  playlist.PlaylistManager - Attempting to load version 2 playlist file [/home/ps4/SDRTrunk/playlist/playlist_v2.xml]
07:20:55.559 INFO  m.d.p.a.P25AudioModule - JMBE audio conversion library [JMBE Audio Conversion Library v0.3.3] successfully loaded - P25 audio will be available
07:20:55.617 INFO  gui.SDRTrunk - starting main application gui
07:22:03.855 DEBUG s.t.u.USBTransferProcessor - R820T 00000001 - buffer overflow cleared - resuming normal processing
07:55:52.621 DEBUG s.t.u.USBTransferProcessor - R820T 00000001 - buffer overflow cleared - resuming normal processing

pc specs: inxi
CPU~Octa core AMD FX-8120 Eight-Core (-MCP-) speed~3110 MHz (max) Kernel~4.4.0-75-lowlatency x86_64 Up~45 min Mem~3110.5/11961.0MB HDD~250.0GB(11.6% used) Procs~273 Client~Shell inxi~2.2.35  
ps4@PS3:~/Downloads/sdrtrunk$ inxi -I
Info:      Processes: 272 Uptime: 47 min Memory: 3126.8/11961.0MB
           Client: Shell (bash) inxi: 2.2.
ps4@PS3:~/Downloads/sdrtrunk$ inxi -A
Audio:     Card-1 NVIDIA Device 0fba driver: snd_hda_intel
           Card-2 Advanced Micro Devices [AMD/ATI] SBx00 Azalia (Intel HDA)
           driver: snd_hda_intel
           Sound: Advanced Linux Sound Architecture v: k4.4.0-75-lowlatency
lsusb 
Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 002: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

Activity Summary - Decoder:P25 Normal (C4FM)
======================================================
SITE h RFSS-SITE:01-42h
------------------------------------------------------
SCCH DOWNLINK 856.587500 [0-893] UPLINK 811.587500 [0-893]  SCCH 2 DOWNLINK 854.137500 [0-501] UPLINK 809.137500 [0-501]
SNDCP DOWNLINK 0.000000 [15-4095] UPLINK 0.000000 [15-4095]
------------------------------------------------------
FREQUENCY BANDS:
 ID:0 BASE:851006250 BANDWIDTH:12500 SPACING:6250 OFFSET:-45000000
 ID:4 BASE:935012500 BANDWIDTH:12500 SPACING:12500 OFFSET:-39000000
------------------------------------------------------
NEIGHBOR SITES: 
NAC:4C0h SYSTEM:4C9h LRA:00h RFSS-SITE:01-08h 
  PCCH: DOWNLINK 0.000000 [1-1930] UPLINK:0.000000 [1-1930] SERVICES: CONTROL CHAN[ ] SERVICES[ DATA REGISTRATION VOICE ]

this is with beta 9 cant seem to get beta 10 to work. have yet to hear any sound while using p52-1
os is skyview linux . going back to linux mint nothing works and no support for this os lol. the only good thing is it showed me your program and your great work :100:

commit 183eb2f7 (2017-02-10): Resolves #165 - P25 decoding channel stops arbitrarily and never recovers.

Issue identified with Patch Group management and circular dependencies that can develop because of stale patch groups.  Refactored all patch group management into a new PatchGroupManager.  Patch groups are now automatically deleted if we don't receive any update messages within 5 seconds.  The PatchGroupAlias was also updated to not roll-up attributes from subordinate patched group aliases when those aliases are also patch group aliases -- this eliminates the circular dependency chaining.

Added IOverflowListener inter [... cut]
PR #177 (2017-02-10): 165 - P25 Channel stops decoding
Resolves #165 issue where P25 Patch Group management was causing a logic error and causing the decoder's dispatcher thread to die.
commit 12ecca45 (2017-02-10): Merge pull request #179 from DSheirer/169-aarch64-usb-support-for-ARM64

Resolves #165 - adds ARM64 (aarch64) linux libusb support
commit 4c27d041 (2017-02-10): Resolves #165 - adds ARM64 (aarch64) linux libusb support