DSheirer/sdrtrunk #168: LIBUSB_ERROR_OTHER ?  (0.3.0 beta 2)

chudgoo, 2017-01-31:
So I ran the latest 0.3.0 beta 2 overnight and came in to see about 20 of these:

19:36:51.505 ERROR s.t.u.USBTransferProcessor - R820T SER#3 - error submitting transfer [LIBUSB_ERROR_OTHER]

Interestingly enough, the application itself doesn't seem affected by it.
This doesn't seem to be a serious issue, but I thought I'd throw it out there.

environment info : This is one of the cheap tuners (RTL2832 SDR/R820T) connected to a USB3 hub, connected to a late 2013 Mac Pro's USB3 port.  (2.7 GHz 12-Core Intel Xeon E5 / 64GB RAM)

davidmckenzie, 2017-01-31:
USB3 has been known to cause issues with the dongles - do you have a USB2 hub you can try? Preferably one with external power :)

chudgoo, 2017-01-31:
I've moved it over to a USB 2 port on the thunderbolt display connected to the Mac Pro for now.  
The weird thing is the is the exact same physical setup I've been using (at my 'day time testbed') for over a year now and this is the first time I've seen that particular error.  
Also, it popped up overnight while no other I/O heavy operations were happening and no other devices were on that hub.  Kind of odd.   I'll let it run again tonight while directly connected and see what happened in the morning.

chudgoo, 2017-02-02:
Just an update... I tried it with the same sdr on a usb2 hub and didn't see the issue again.
Then I put two dongles on the same usb3 hub connected to a usb3 port on the mac pro and still haven't seen it come back.   Not sure what happened there...

chudgoo, 2017-02-02:
It looks like I spoke too soon... or wasn't paying attention...
The following is while using dual tuners on the same USB3 hub.  I'll now put them each on their own USB2 port and run over the weekend.

14:01:10.603 INFO  gui.SDRTrunk - starting main application gui
14:54:57.507 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - error submitting transfer [LIBUSB_ERROR_OTHER]
18:01:24.509 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - error submitting transfer [LIBUSB_ERROR_OTHER]
19:02:30.506 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - error submitting transfer [LIBUSB_ERROR_OTHER]
20:03:34.507 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - error submitting transfer [LIBUSB_ERROR_OTHER]
21:04:38.508 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - error submitting transfer [LIBUSB_ERROR_OTHER]
21:05:44.057 DEBUG s.t.TunerChannelSource - Channel [407251250] buffer overflow cleared - resuming normal processing
22:05:42.506 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - error submitting transfer [LIBUSB_ERROR_OTHER]
22:07:01.479 ERROR r.w.RealBufferWaveRecorder - Error starting real buffer recorder
java.nio.channels.ClosedByInterruptException: null
	at java.nio.channels.spi.AbstractInterruptibleChannel.end(AbstractInterruptibleChannel.java:202) ~[na:1.8.0_112]
	at sun.nio.ch.FileChannelImpl.write(FileChannelImpl.java:216) ~[na:1.8.0_112]
	at record.wave.WaveWriter.open(Unknown Source) ~[sdrtrunk.jar:na]
	at record.wave.WaveWriter.<init>(Unknown Source) ~[sdrtrunk.jar:na]
	at record.wave.WaveWriter.<init>(Unknown Source) ~[sdrtrunk.jar:na]
	at record.wave.RealBufferWaveRecorder.start(Unknown Source) [sdrtrunk.jar:na]
	at record.RecorderManager.receive(Unknown Source) [sdrtrunk.jar:na]
	at record.RecorderManager.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.audio.P25AudioModule.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.audio.P25AudioModule.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.Decoder.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.Decoder.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.Decoder.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageProcessor.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageProcessor.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer.dispatch(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer.access$300(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer$P25MessageAssembler.checkComplete(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer$P25MessageAssembler.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer.receive(Unknown Source) [sdrtrunk.jar:na]
	at module.decode.p25.P25MessageFramer.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at dsp.psk.QPSKPolarSlicer.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.psk.QPSKPolarSlicer.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.psk.LSMDemodulator$GardnerSymbolTiming.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.psk.LSMDemodulator.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.gain.ComplexFeedForwardGainControl.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.complex.ComplexBufferToStreamConverter.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.complex.ComplexBufferToStreamConverter.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.fir.complex.ComplexFIRFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.fir.complex.ComplexFIRFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.broadcast(Unknown Source) [sdrtrunk.jar:na]
	at sample.Broadcaster.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.halfband.complex.HalfBandFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.halfband.complex.HalfBandFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.fir.complex.ComplexFIRFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.fir.complex.ComplexFIRFilter_CB_CB.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.complex.ComplexToComplexBufferAssembler.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$Output.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.decimator.ComplexDecimator.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$Stage.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$DecimatingStage.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.decimator.ComplexDecimator.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$Stage.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$DecimatingStage.receive(Unknown Source) [sdrtrunk.jar:na]
	at sample.decimator.ComplexDecimator.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$TwoStage.receive(Unknown Source) [sdrtrunk.jar:na]
	at dsp.filter.cic.ComplexPrimeCICDecimate$DecimatingStage.receive(Unknown Source) [sdrtrunk.jar:na]
[... cut ...]
11:45:28.090 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - unable to clear device endpoint halt
11:45:28.095 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - error submitting transfer [LIBUSB_ERROR_OTHER]
11:45:28.100 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - transfer error [TRANSFER ERROR (1)] transferred actual: 0
11:45:28.100 ERROR s.t.u.USBTransferProcessor - R820T 00000001 - error submitting transfer [LIBUSB_ERROR_OTHER]
11:45:28.100 WARN  s.t.u.USBTransferProcessor - R820T 00000001 - all transfer buffer processing is stopped

DSheirer, 2017-02-14:
Are you still seeing this issue when running 2 tuners?

closed 2017-12-24 by DSheirer