DSheirer/sdrtrunk #169: FR: aarch64 support?  libusb4java

chudgoo, 2017-01-31:
Hello again~
I have been sent a version of libusb4java that I was told was compiled for aarch64, but I'm not entirely sure how to turn the .so file into a usable .jar. 

I'd love to see sdrtrunk supported on 64 bit ARM architectures.  Is this process something you're familiar with?

https://www.dropbox.com/s/urk5eqn5as5drf0/libusb4java.so?dl=0

DSheirer, 2017-01-31:
I'll get this bundled up and included in the project.

DSheirer, 2017-02-10:
aarch64 native library now included in master branch.

closed 2017-02-10 by DSheirer

chudgoo, 2017-02-10:
Awesome!  I will start testing it tonight on the ODROID C2.  
I'm hoping that with the reduced audio processing load and various other changes you've made recently, that this little $40 SBC -may- be able to replace my primary streaming box, a 2014 Macbook Pro.  I'll chime in with results on this thread later.  Thanks @DSheirer !

spinnaker813, 2018-03-18:
Wondering if there is an update for the ODROID C2, looking to use one on a project I am working on. How capable is it at decoding a large P25 Phase 1 system