DSheirer/sdrtrunk #172: Allow user to limit tuner frequency range

DSheirer, 2017-02-02:
Provide capability for user to rescope the tuner's min/max frequency range.  Update tuner's editor panel to allow them to select a min/max range that is a subset of the tuner's true min/max range, or reset that range to the true min/max range.

This allows users that have tuners connected to specific antenna configurations to limit tunable frequency range.

frameshift18, 2017-02-02:
Would love this feature, often the control channel frequency ends up at the far edge of the SDR's range and the signal drops out rendering the program unable to track.

ImagoTrigger, 2017-02-28:
this potentially is related to #119 you could knock em both out as it seems they deal with similar parts of the src (i.e. freq range = 0mhz-0mhz for serial number 0000002)

davidmckenzie, 2017-03-01:
Would this be limiting the range of frequencies that can be the centre frequency of the tuner, or a range of frequencies for channels that can be used?

If undecided, personally I'd go for the former. As @frameshift18 mentioned, a common problem is that the centre frequency is changed to follow a voice channel, resulting in losing the control channel. By setting a range for the centre frequency, it would allow the tuner to lock on to a voice channel at the edge of the range but not change to a point where CC is lost.

I might be explaining this bad, so here's an example:

CC is 416.25MHz, one VC is 415MHZ, sample rate is 1.44MHz. When that VC is active, the tuner centres on 415.5MHz, putting the CC right at the edge of the spectrum. If it centres to 415.6MHz instead, both channels are in the receivable range.

If the tuner range we're setting is the centre frequency, then I could set this tuner to centre no lower than 415.6MHz, thus never losing CC. If we're setting the range of channels to tune, I'd have to block that tuner from using that particular VC.

The only downside is that this would be pretty damn confusing for users, and may run counter to what you envisaged. :)

hans109h, 2019-12-13:
A method of this that would work for me would be to restrict tuner use to tuners selected in the channel source list.

DSheirer, 2024-02-11:
Seven ... years ... later ...

GTR8000, 2024-02-11:
AWESOME!!! No more creating dummy channels to keep a range centered!

PR #1850 (2024-02-11): #172 Tuner Supports Adjusting Min/Max Frequencies
Enhances tuner editors to allow users to specify minimum and maximum frequency values to more accurately control tuner behavior.

Closes #172
closed 2024-02-11 by DSheirer