DSheirer/sdrtrunk #175: Support stereo streaming

davidmckenzie, 2017-02-06:
I know this has been discussed before, but I couldn't find an issue for it. Pretty sure it was a milestone for a distant future feature? Either way, just had an idea of how this could be implemented in a fairly neat way that retains the core appeal of sdrtrunk for streaming - never missing a message.

This could be implemented through a checkbox on the streaming configuration for stereo mode. A separate queue is established for left and right channels, both of which are processed in order.

Aliases can have a tag added to only stream on left or right channel. If no channel tag is present then the shorter queue (at the time of recording the temporary file) is used. Once the untagged alias is allocated to a channel, any further messages from that alias are added to that same queue, until the queue reaches 0. This would prevent recordings from playing out of order, while still providing enough flexibility to avoid excessive queuing.

The broadcasting processer looks at both queues: 
* If one queue has a message but the other doesn't, stream immediately with 0s on the empty channel.
* If both queues have one message, immediately stream both files (similar to using sox to combine two mono files into one stereo file). Append 0s to the shorter of the two messages if required.
* If one queue has one message, but the other queue has more than one message, measure the combined length of the messages in the longer queue until you find how many will fit. E.g. left channel has 1 message of 1 minute length, right channel has 3 messages of 25 seconds length; stream 2 messages from right channel with the 1 from left
* If both queues have more than one message, use the same logic as above, finding how many messages would fit in the space of the longest of the two messages.

If the last two logic conditions cause performance degradation, then we could shorten the max message length and only go with the first two. If the files were split into 10s chunks and you had someone rambling on with a 2 minute long message on the left channel, it wouldn't cause the right channel to queue for a couple of minutes.

This would also treat #164 for some people - if the two patched channels were each assigned left or right, then the duplicate messages would (maybe) stream simultaneously, with a little echo.

ImagoTrigger, 2017-03-01:
im messing around with this using a custom player outside of SDRTrunk (still utilizing SDRTrunk to recording .wavs) to find a good combination of streaming mono to both speakers when its slow and using split left and right when its busy / falling behind.

ImagoTrigger, 2017-03-02:
After lots of trials I've finally got a good setup going (on Windows).  

You just have to turn SDRTrunk headless mode on (#92), disable your stream sources in SDRTrunk, also disable it's mixer (#213) and turn audio recordings on.  

Now get yourself an Icecast server with the Altacast source provider, then run these 2 little perl scripts to drive the mix:
[stream.pl.txt](https://github.com/DSheirer/sdrtrunk/files/812781/stream.pl.txt)
[stream2.pl.txt](https://github.com/DSheirer/sdrtrunk/files/812782/stream2.pl.txt)

Lastly, you listen (and watch) with the http://www.foobar2000.org/ player.

The result is a stable listening station that displays alpha tags on time, every time.  (don't use shoutcast or VLC)

aright now it's just a prototype but you start stream.pl first then stream2.pl.  if the station is 10 calls behind it will kick in and it's 50/50 luck with a little logic thrown in for good measure.  there are 2 flat files that contain the to and the from alpha tag lookups.

[image]

and sure here is my startup script in case the machine reboots it all comes back automatically:
[start_scanner.pl.txt](https://github.com/DSheirer/sdrtrunk/files/812814/start_scanner.pl.txt)