DSheirer/sdrtrunk #181: Latest build - buffer overflow cleared

davidmckenzie, 2017-02-10:
Hi @DSheirer - thanks so much for the latest build, it's extremely timely as my streams will be getting a workout today :)

Not sure if this is an issue as it doesn't seem to be affecting reception so far, but figured it's worth posting in case this is not expected behaviour. Seem to have had a few of the buffer overflow messages shortly after starting the app:

```
07:13:52.408 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
07:13:57.024 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
07:13:59.163 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
07:14:05.195 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
07:14:07.326 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
07:14:34.766 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
07:14:37.882 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
07:14:44.160 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
07:14:46.290 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
07:15:18.483 DEBUG s.t.u.USBTransferProcessor - R820T 0100 - buffer overflow cleared - resuming normal processing
```

The 0100 device is the one I have dedicated to voice channel following @ 2.4M sample rate. Haven't had any more errors for a few minutes now so will continue to monitor and try out a few different things.

davidmckenzie, 2017-02-10:
Further observation - started getting the duplicate calls on patched TGIDs. There are no duplicate events, no duplicate channels like my previous issue. This is just a simple patched TGID broadcasting twice :(

TGID 4E23 was patched with 4E2A. Both TGIDs had an alias set up with a broadcast channel and audio priority. I've set 4E2A to 'do not monitor' and removed the broadcast channel, will see if that stops the dupe messages.

davidmckenzie, 2017-02-10:
Logs from the affected tower:

```
START_DATE,START_TIME,END_DATE,END_TIME,DECODER,EVENT,FROM,FROM_ALIAS,TO,TO_ALIAS,CHANNEL,FREQUENCY,DETAILS
'20170211','073320','','',','P25 Phase I','Patch Group Add',,,,,'null','0','PATCH GROUP:2794 [278B]'
'20170211','073321','','',','P25 Phase I','Patch Group Add',,,,,'null','0','PATCH GROUP:4E23 [4E2A]'
'20170211','073329','','',','P25 Phase I','Register','1F020D',,'753D','RFS','null','0','REGISTRATION:ACCEPT SITE: 02-35'
'20170211','073329','','',','P25 Phase I','Patch Group Add',,,,,'null','0','PATCH GROUP:7534 [753D]'
'20170211','073343','','',','P25 Phase I','Data Call',,,'1EA441',,'4-804','417500000','PACKET'
'20170211','073358','','',','P25 Phase I','Data Call',,,'1EA441',,'4-404','415000000','PACKET'
'20170211','073413','','',','P25 Phase I','Data Call',,,'1EA441',,'4-764','417250000','PACKET'
'20170211','073428','','',','P25 Phase I','Data Call',,,'1EA441',,'4-1038','418962500','PACKET'
'20170211','073358','20170211','073535','','P25 Phase I','Data Call',,,'1EA441',,'4-404','415000000','PACKET'
'20170211','073535','','',','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073535','','',','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073537','','',','P25 Phase I','Register','2191C9',,'521D','Other','null','0','REGISTRATION:ACCEPT SITE: 02-35'
'20170211','073535','20170211','073540','','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073535','20170211','073541','','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073541','','',','P25 Phase I','Patch Group Call','1EB909',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073541','','',','P25 Phase I','Patch Group Call','1EB909',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073541','20170211','073544','','P25 Phase I','Patch Group Call','1EB909',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073544','','',','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073544','','',','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073544','20170211','073555','','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073544','20170211','073556','','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073556','','',','P25 Phase I','Patch Group Call','1EB909',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073556','20170211','073600','','P25 Phase I','Patch Group Call','1EB909',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073600','','',','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073600','20170211','073604','','P25 Phase I','Patch Group Call','1EE343',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073604','','',','P25 Phase I','Patch Group Call','1EB909',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073604','','',','P25 Phase I','Patch Group Call','1EB909',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
'20170211','073604','20170211','073609','','P25 Phase I','Patch Group Call','1EB909',,'4E23','eG021 CHIFLEY','4-404','415000000','PRI:0 PACKET'
```

DSheirer, 2017-02-10:
The 'buffer overflow cleared' happens when the tuner is producing samples but the downstream processing is not keeping up.  It enters an 'overflow' state where it throws away samples until processing catches up.

As for streaming, I'm working on that now.  I've replicated the repeating audio calls streaming to my local server, just trying to figure out what's causing it.

davidmckenzie, 2017-02-10:
Thanks dude :) No further buffer errors but I notice that I get a few of them every time I close/re-open the app. I'm guessing it's just a little too verbose at the moment?

Further observation on the streaming if it helps - having the 'partner' of the patch TGID set to 'do not monitor' made the patch group calls marked as 'do not monitor'.

I.e. I had 4E2A set to 'do not monitor' but 4E23 had a high priority. Calls to the patch group of 4E23 were not monitored.

DSheirer, 2017-02-10:
For the 4E2A (do not monitor) and 4E23(high audio priority) ...

I designed it so that the patch group assumes the highest audio priority from all of the patched groups and from the patch group alias (if one exists).  -1 is do not monitor, 0 is channel selected, and 1-100 follow after that.  So, if one of the groups is 'do not monitor', then the patch group will be considered do not monitor since the -1 priority being the lowest value (ie highest priority) will percolate to the top.

I could go either way on this: 1) use the highest (listenable) priority or, 2)leave it the way it is and if any of the pachted groups has do not monitor, the patched group doesn't get monitored.

Thoughts?

davidmckenzie, 2017-02-10:
My vote would go towards 1. Lets say for example that you have "County A" and "County B" - you don't care about what goes on in B, but you care about A. When B is patched with A, you'd want to hear it still.

I'm struggling to think of a use-case for 2 - did you have any in mind?

DSheirer, 2017-02-10:
I guess I was thinking along the same lines ... but from the point of view of the 'do not monitor' group, thinking that if you marked it as 'do not monitor', maybe that means do not monitor it under any circumstances.  

Personally, I can see either approach being valid ... and your vote tips the scales :-)

DSheirer, 2017-02-10:
Oh, and I found the duplicate audio bug!

davidmckenzie, 2017-02-10:
You bloody legend, thanks Denny!

So far so good with the new build :)

commit 7c930dee (2017-02-11): Resolves #181 - patch group audio priority will assume the highest (listenable) audio priority and override a 'Do Not Monitor' setting if any of the other aliases have an audio priority defined.
commit 861203c3 (2017-02-11): Merge pull request #184 from DSheirer/181-patch-group-audio-priority

Resolves #181 - patch group audio priority will assume the highest (l…
closed 2017-02-11 by DSheirer