DSheirer/sdrtrunk #197: Audio recordable alias

davidmckenzie, 2017-02-13:
Hi Denny!

Would it be possible to have an alias tag to enable recording of a single alias? Basically the opposite of the non-recordable tag. I'd like to only record audio on one talkgroup (to identify RIDs), but doing so would mean adding the non-recordable tag to 20+ other aliases.

Thanks!
Dave

CGUTH, 2017-02-14:
One option is you can highlight multiple aliases and set the Dot Not Record tag.

davidmckenzie, 2017-02-14:
Omg, how did I not know you could select multiple aliases :D

[image]

CGUTH, 2017-02-14:
I just figured it out by accident 2 days ago, I love it.

DSheirer, 2017-02-14:
I probably need to get the user's manual updated this weekend.

DSheirer, 2017-02-17:
Did the multi-select and 'Non-Recordable' work out ok for you?

davidmckenzie, 2017-02-18:
Yup! Thanks both!

closed 2017-02-18 by davidmckenzie