DSheirer/sdrtrunk #198: Audio recording file handles left open

ImagoTrigger, 2017-02-15:
When you tell sdrtrunk to write .wavs to /recordings/ it seems that it will sometimes not close all the files when done writing.

ImagoTrigger, 2017-02-17:
[image]

ImagoTrigger, 2017-02-28:
going to try to repro on XP/Linux just to make sure it's not my setup...  probably not closing the filechannel....poison pills... ugh.... *headache*

ImagoTrigger, 2017-02-28:
[image]

ImagoTrigger, 2017-03-01:
think i found this one too...gotta let it run for a while to be sure

ImagoTrigger, 2017-03-01:
[image]

ImagoTrigger, 2017-03-01:
well my change made it more atomic lol but not fixed... still leaving handles open regardless of the platform

ImagoTrigger, 2017-03-01:
[image]

ImagoTrigger, 2017-03-01:
fixed it.....

[atomicfiles-patch.txt](https://github.com/DSheirer/sdrtrunk/files/811232/atomicfiles-patch.txt)

DSheirer, 2017-03-03:
@ImagoTrigger thanks again for finding the problem and figuring out the corrective patch!!

closed 2017-03-03 by DSheirer

ImagoTrigger, 2017-03-05:
thanks for fixing it correctly ;-)

DSheirer, 2017-03-05:
Team work!