DSheirer/sdrtrunk #200: Add an option to force "Do Not Stream" when traffic is encrypted

ImagoTrigger, 2017-02-15:


DSheirer, 2017-02-17:
This will depend on enhancing the P25 message error correction routines to get consistent indicators of encrypted audio.  Currently, errors in decode produce false encryption triggers.

Once the P25 error correction routines are resolved, the encrypted audio will be automatically suppressed.

robhancock, 2017-05-24:
This is likely the same issue as #244. The problem seems to be that LDU1 frames are always falsely detected as unencrypted and thus it tries to play out half of the encrypted frames that are received.

DSheirer, 2017-05-28:
This issue should be resolved with the patch for issue #244 in version 0.3.0 Beta 12.

There is a slight possibility that the HDU or LDU2 messages may be corrupted and incorrectly indicate unencrypted when the call is actually encrypted ... this would be a result of not error-correcting the messages.  In this rare instance, the encrypted audio may playback. 

Please let me know if you are still getting encrypted audio playback?

Thanks,
Denny

closed 2017-05-28 by DSheirer

vsboost, 2017-09-08:
Occasionally i hear encrypted traffic using  v0.3.0 beta14