DSheirer/sdrtrunk #204: Receiving data but not audio playback

youcangetme, 2017-02-18:
Hello! Thank you for the cool software.

I have tried to set the audio outputs but no luck. My system is some generic surround sound from RealTek. I can see the reception in the spectrum analyzer so I know the app is talking to the SDR. I do have a HDMI monitor with audio but I have also used headphones with the same results, not sure if that matters. I gave up on any meaningful audio coming out of Java a long time ago so not sure how to help you debug this.

davidmckenzie, 2017-02-18:
Anything in the console output?
Anything in the events and messages panels for each channel?
Have you compiled + added JMBE? https://github.com/DSheirer/sdrtrunk/wiki/JMBE

youcangetme, 2017-02-18:
Ah! OK, so I didn't have the channel enabled. Makes since.

Sorry for the alarm. Maybe a brief tutorial would help?

closed 2017-02-20 by youcangetme