DSheirer/sdrtrunk #210: Playlist saving all the time...

ImagoTrigger, 2017-02-28:
sdrtrunk saves the playlist even when the playlist does not change.  this is a noticeable problem when my playlist xml file size is 10MB.  ive just gone ahead and commented out the schedulePlaylistSave() calls (besides running my entire setup from RAM)....

ImagoTrigger, 2017-02-28:
hint: it's only due/with p25 channelmap?

DSheirer, 2017-03-01:
It's only supposed to save the file at a max rate of once every 2 seconds when there have been one or more changes to aliases, channels, channel maps, or streaming configurations.

There can be a noticeable delay between when the change happens until the playlist is actually saved.

Are you seeing additional/recurring saves beyond the times when you're making changes?

DSheirer, 2017-03-01:
Second question ... I'm planning on supporting multiple playlists in the near future, so that you can create custom playlists and don't have to have everything jammed into a single playlist.  Would that allow you to reduce the size of your playlist, if you could split it into multiple playlists and select one at runtime?

ImagoTrigger, 2017-03-01:
i figured it out i tihnk

BROADCASTER_QUEUE_CHANGE is triggering it. 
[image]

ImagoTrigger, 2017-03-01:
here is da fix

[image]

ImagoTrigger, 2017-03-01:
as for the second question it is a non issue at this point ;-) my playlist does not need to be so big... would be nice to be able to run a separate playlist on each tuner from one instance of sdrtrunk, why else have a playlistmanager manager...

DSheirer, 2017-03-03:
@ImagoTrigger thanks very much for tracking this down!

PR #214 (2017-03-03): Resolves issue with the playlist saving all of the time
Resolves #210 issue with the playlist saving all of the time due to audio streaming broadcast events.
closed 2017-03-03 by DSheirer