DSheirer/sdrtrunk #220: Log Data Calls

sysrun, 2017-03-06:
Is it possible to log the contents of data calls for P25 somehow ?

DSheirer, 2017-03-06:
I don't have an option to log the decoded messages from a traffic channel, but that's a good idea.

The decoded messages will show you the SNDCP data message contents if the protocol is IP/UDP or IP/TCP.

Is this what you're asking for, or are you looking for something like a packet capture?

sysrun, 2017-03-06:
That would be awesome :)

DSheirer, 2017-03-06:
-Add option to gui logging configuration editor to select logging for traffic channels.
-Update TrafficChannelManager to create a logging configuration instance that contains the traffic channel logging tag and add to the traffic channel configuration prior to starting via the ChannelProcessingManager.

DSheirer, 2017-12-24:
Resolved under #299

closed 2017-12-24 by DSheirer