DSheirer/sdrtrunk #227: Feature: Astrometa DVB-T/T2 receiver support

Forst, 2017-03-22:
These receivers work well with for example gqrx. There is already support in librtlsdr, it was added more than three years ago (steve-m/librtlsdr@aefd8b7d58c4735b19110b5f54de289e8dd931f5). I have one myself, so I can test if this ever gets implemented.

Vendor ID `0x15f4`, product ID `0x0131`.

am-silex, 2024-04-23:
It's interesting that support for this tuner hasn't been implemented since 2017 and nobody chimed in with solution.

TheDarkXli, 2026-02-22:
Hi — I’m still running into the same problem described here with generic RTL2832U + R828D dongles (not RTL‑SDR Blog or V4 models). In my setup on macOS with sdrtrunk nightly, I see frequent “Unable to initialize R828D” errors and the tuner never stabilizes: 
```
INFO … PLL is not locked. Increasing VCO current
ERROR … Unable to start tuner [RTL‑2832] ‑ error: Can't claim interface on USB tuner ‑ LIBUSB_ERROR_ACCESS
```

Using `rtl_test` I confirm the dongle works but shows a large frequency error (~+154 ppm) and persistent “[R82XX] PLL not locked!” warnings even after warmup. These sticks can be made usable with manual ppm correction and lower sample rate but the core driver always fails to initialize reliably in SDRTrunk with trunking workflows.

Here is the `rtl_test -p` output showing ~+154 cumulative PPM and unstable PLL behavior:
```
real sample rate: 2048334 current PPM: 163 cumulative PPM: 156
real sample rate: 2048312 current PPM: 152 cumulative PPM: 155
```
… repeated similar values …

This suggests the tuner PLL + frequency correction flow in sdrtrunk’s RTL driver isn’t handling these devices the way librtlsdr itself does. Since librtlsdr has had R828D support for years, it might help to align SDRTrunk’s driver logic with the newer code paths or add explicit R828D handling to avoid premature initialization failures.

Thanks for maintaining this project!

DSheirer, 2026-11-07:
closing as implemented with the new jRTL driver library.

closed 2026-11-07 by DSheirer