DSheirer/sdrtrunk #244: Should not try to play P25 encrypted audio

robhancock, 2017-05-03:
When monitoring a P25 system, it appears that the app is trying to play back all P25 audio even it is encrypted, resulting a bunch of gibberish. The call seems to be getting detected as encrypted (i.e. under Events for the control channel, the details shows the call as ENCRYPTED) but it still seems to be trying to play it anyway.

robhancock, 2017-05-03:
I noticed that when it's playing encrypted audio it also cuts in and out. It appears from the Messages output that the LDU2 messages are being detected as encrypted but LDU1 are not. Here's an example of one of the LDU1 messages that it seems to think is not encrypted but is:

`21:08:22	P25 Phase 1	**f	NAC:4E3 LDU1  VOICE LSD:6183 GRP_VCH_USER     ENCRYPTED CHANNEL FROM:008190 TO:07D1	010011100011010110101100111011110101101100000110000100100111100100101101000001001101011000000100100001100000011110100010000011110010101000011001001101111101011001100101011111101001010100101101100111100111100100010001100001101110001000000110110111000100001011100000111011000100111001110111000001110000001101111000101100011000111111010111011110100001001100000000000000000000000001110000010001110011001101100001110111001100100001001110010101101010011111001100110000001100101110001100111010000000110000001001101001100000100110010000011001010000000000000000000001111111100100010001010010000011000010010010111100010010010001101010100101100000001101011110110100111100010110000001100111010010001111010111100011101100100101100111000000000000100010110001100100010000110100001110110001110011111001011101111110001001001110010011111001011010100011110110100100101010001111000001011001000111001000011011000010000001001000111100111001101010101111110110111111010100011110101010110000111111111100011010011001111101111011000110011101010110111111110010010001011010001111011010111111110110010000000010010111001010111010001100110001110111011100100000100111110001101100100101010001110110001001111111001001100111110100110111111101101111110000001000011010001101011101101111110111010101000110111011111110110001011101011001101010111101001101101001100000110000110000101110100111001111000111111010111111011101000111101101111010110101000111101100111100101000001000111100111010111011001001100001100100011000001100000101111100100101100010000101110001100101001010101001011101110100000100110110011010101100101000000010111000101001111110111000110111000011100101111100`

robhancock, 2017-05-03:
The code is using an encrypted flag in the LDU1 message to determine if the LDU1 message content is encrypted. I don't have a copy of the P25 standard to verify whether that makes any sense but from what I'm seeing it doesn't seem to work in any case. From what I can tell, other apps like DSD seem to parse LDU1 together with LDU2, which has all the encryption details in it.

I did a quick hack to store the encryption state of the last LDU2 message and use that state when processing the next LDU1 message. That seems to work fairly well, except for a brief burst of gibberish when going from an unencrypted call to an encrypted one, where the first encrypted LDU1 message gets played incorrectly. Really it seems like it should be pairing up LDU1 and LDU2 messages before trying to play them out rather than playing both types individually.

ImagoTrigger, 2017-05-24:
@robhancock can you make a patch file with your hack in it?  i was trying to do something similar see #200

robhancock, 2017-05-24:
Patch for my hack is attached.

I found some references in this paper: https://online.wsj.com/public/resources/documents/p25sec08102011.pdf that allude to the "protected" bit in the LDU1 as referring to the link ID fields in the link control word, not to the voice payload itself. If that is indeed the bit that the current code is checking in LDU1 to check for encrypted voice, that's definitely incorrect: 

> The ID fields in the LCW can be optionally encrypted, but whether they are actually encrypted is not intrinsically tied to whether encryption is enabled for the voice content itself (rather it is indicated by a “protected” bit flag in the LCW).
> We found that in every P25 transmission we captured, both in P25 transmissions sent from our equipment and from  encrypted  traffic  we  intercepted  over  the  air,  the LCW  protection  bit  is  never  set;  the  option  to  encrypt the LCW does not appear ever to be enabled, even when the voice traffic itself is encrypted.  That is, in both Motorola’s  XTS5000  product  and,  apparently,  in  virtually every other P25 radio in current use by the Federal government, the sender’s Unit Link ID is always sent in the clear, even for encrypted traffic.

[sdrtrunk-encryptedaudio.txt](https://github.com/DSheirer/sdrtrunk/files/1024257/sdrtrunk-encryptedaudio.txt)

closed 2017-05-28 by DSheirer