DSheirer/sdrtrunk #248: Occasionally locks up

mlbrianr, 2017-06-01:
I am monitoring one P25 system.   At various times, I look at the app and see that it's locked up, and I always see one of the traffic channels in "TEARDOWN" status.   I can click on tabs but the waterfall graph is not moving and no decoding is occurring.   Is there any further information I can provide on this which might be useful for troubleshooting?  Thanks..

[image]

davidmckenzie, 2017-06-01:
When the waterfall's not moving, that usually indicates that your USB device has died or been disconnected somehow.

What does the console log say?

How many tuners do you have connected?

chudgoo, 2017-06-22:
I too have been seeing this with beta 12.  Except in my case, the waterfall still moves, but I do start seeing "**F" and "*FF" in the control channel / messages pane / error check column when it happens.  Also I tend to see many TEARDOWN and OVERFLOW messages in the 'now playing' pane and no audio is produced.

DSheirer, 2017-06-26:
Sorry for the delayed response ... it's been a busy summer.

When it locks up, the first place to check is the application log:  [user directory]/SDRTrunk/logs/sdrtrunk_app.log

Look for anything that mentions out-of-memory errors, or send it to me and I can see if there are any issues.

The issue with the TEARDOWN state not clearing may be an indication of an error in that section of the code.  I noticed that I'm catching general Exceptions, but not catching Throwable errors.  I'll change that in the code so that it will log if/when there is a code error that may be causing the state processing thread to die and not change the state from TEARDOWN to IDLE.  If it gets stuck in the TEARDOWN state, that channel/frequency will no longer produce audio until the application is restarted.

edbir1, 2018-04-24:
I am running 3.4.beta.2 on a rpi3+ using 2x rtlsdr.com dongles monitoring a P25 phase 1 system.
I minimize the gui window asap to keep cpu usage in an acceptable range.  I am currently attempting to see how to adjust the tuner bandwidth to try to maximize the number of channels that will fit into the tuner bandwidth and not crash the program.

When I set the tuner bandwidth on the dongle which listens to the control channel to 2.4mhz, so that it can also hear one of the non control channels, I start to get out of memory errors:

09:14:39.917 ERROR i.g.d.s.t.TunerChannelSource - Code error encountered during decimation process - channel thread will probably die  [212MB/212MB 99%]
java.lang.OutOfMemoryError: Java heap space
09:14:40.730 ERROR i.g.d.s.t.TunerChannelSource - Code error encountered during decimation process - channel thread will probably die  [212MB/212MB 99%]
java.lang.OutOfMemoryError: Java heap space

bx-api, 2020-06-02:
Im experiencing the same issue (application freezes showing a channel in TEARDOWN state).

Also using it on P25 phase 1 simulcast with 2 RTL-SDR.com BLOG V3 dongles set to 2.4Mhz

closed 2022-11-26 by DSheirer