DSheirer/sdrtrunk #249: Overflow/teardown lockup...automated kill/relaunch?

chudgoo, 2017-06-23:
I've been having quite a few problems with 'overflow' and 'teardown' crashes and was trying to get a script going that will detect this condition, kill and relaunch when detected.  Even a cron job to kill sdrtrunk every hour or something...  Maybe this is missing the point, and I need another method to troubleshoot this issue.

My environment is two rtlsdrs on a macbook pro with USB3 ports.  I tried both off one port with a USB2 hub. I tried direct connect of each to a USB3 port. I've tried sdr>usb2 hub>usb3 port.  It doesn't seem to make a difference.  The logs don't seem to point me in any solid direction.
I know the Issues section isn't the place for this sort of thing, and if there is, could someone direct me to it please?

DSheirer, 2017-06-26:
I just opened ticket #250 for the channel getting stuck in the TEARDOWN state.  I'll fix this and put out a beta 13 release so that with your help we can try to diagnose what's causing the state monitor thread to die and leaving the channel stuck in the TEARDOWN state.  

I'm traveling for work this week, but I'll get a new beta out as soon as possible.

DSheirer, 2017-07-01:
I published a test release for troubleshooting this issue:
https://github.com/DSheirer/sdrtrunk/releases/tag/v0.3.0-beta.12.idle.state.test

Could you please run this version until you see a channel get stuck in the TEARDOWN state and then send/attach the application log file?  I'm trying to determine if what you're seeing is caused by an error that is killing the channel state monitoring thread.  Thanks!

chudgoo, 2017-07-01:
I will test it later today. Thanks Denny, I'll let you know how it goes.    Also... A fun observation on the topic, out of desperation/frustration i went and threw SEVEN tuners off one USB3 hub feeding into a mac pro (12 xeon cores/64GB ram) and haven't had a single issue with it. Same beta 12.  Very odd.   https://scontent-dft4-1.xx.fbcdn.net/v/t35.0-12/19650174_10213794713081294_111143689_o.jpg?oh=9976c423b4b93ef749bc9f7bca4dd2b6&oe=595A0699

chudgoo, 2017-07-01:
> beta.12.idle.state.test

Will do. Thanks Denny.

chudgoo, 2017-07-01:
Currently running on two machines of vastly different specs. It has been roughly 2 hours and so far not a single teardown hang. Will report back later if there are any issues.

chudgoo, 2017-07-02:
So we're on ~12 hours of two systems running this version with identical (software) configs on very different hardware.... Not a single teardown/idle error in sight.    Whatever you changed seems to have worked.  :)

chudgoo, 2017-07-03:
Well, it took more than 50 hours, but it finally happened.  Here's the last sdrtrunk_app.log file.

[sdrtrunk_app.log.txt](https://github.com/DSheirer/sdrtrunk/files/1120702/sdrtrunk_app.log.txt)
[image]

chudgoo, 2017-07-04:
Not sure if it's relevant, but when this last hang occurred, I was only using three tuners (instead of 7).
Are there any deeper debug level logs, or flags to launch with that provide super verbose logs?

davidmckenzie, 2017-07-04:
@chudgoo curious - what framerate are you running the waterfall at? That banding of frequencies mirrored across the waterfall is what I see whenever I put the fps below 20.

chudgoo, 2017-07-05:
Looks like I had it at 14FPS and 4096 FFT width.  Going to try 30 today and see how it goes.

Also, fun story... at our towns' fireworks display last night we had something like 300 calls (accidents, DUIs, stalled traffic for 2 hours, injured 17 year old with homemade PVC pipe bomb, dumpster fire, tree fire, brush fire, etc) and throughout all of that action, SDRTrunk didn't miss a beat. I was listening to the stream all from 4PM-1AM and it's still running now.   

I'm starting to wonder what "the" condition is that causes the hangups, because it doesn't occur very often...especially with the new test build.

chudgoo, 2017-07-08:
I'm curious... are there things in the various logs that could be correlated to find the cause?  I ask because it ran for multiple days (including one incredibly busy one) and ran 100% great, but then in the last two days seems to crash about every 3 hours. No idea what changed on the system,. but my config and build of sdrtrunk did not.  Is there a chance that encrypted calls could be behind this?  We rarely have them here.... it's literally the only external factor I can think of that's sporadic and rare like these crashes seem to be

DSheirer, 2017-07-09:
@chudgoo is the program crashing (locking up) or is one or more of the channels getting stuck in the TEARDOWN state?  Usually a crash/lock up will be an out-of-memory situation.  A channel stuck in a perpetual TEARDOWN state will likely be a threading issue and there may not be any log entries. I'm trying to recreate the TEARDOWN state issue here on my test machine, but no luck yet.

DSheirer, 2017-07-09:
I found/fixed an issue where 'overflow' log messages were being logged erroneously when you are using 2 or more tuners.  I also tweaked the buffer handling a bit to improve processing if/when the tuner gets into an overflow state.  There was also bug where one of the components was not deregistering correctly, but I don't think that would have contributed to channel state lockups -- it would only have prevented the LIBUSB processor from shutting down during idle situations.  These changes are in issue #252 and are now merged into master.

I'm going to keep running/testing the application for a while with a 2-tuner setup to see if I can recreate the stuck TEARDOWN state situation.  At some point I'll push a new test build for you to try.

chudgoo, 2017-07-11:
Hi Denny, sorry I didn't see this until just now. The application itself remains running. You can click around the various panes/tabs, but the now playing pane lists (usually two) red TEARDOWN lines and call processing/audio stops.  Should I try using the new master?

chudgoo, 2017-07-11:
So this one was pretty odd... It occurred after only 17 minutes of running. 
Also, I had the waterfall paused, and only one channel enabled. Only two tuners this time as well.

One strange thing I hadn't noticed before was the TEARDOWN channel (which isn't the CC, btw) continues to strear messages under the Messages tab.

[image]

[image]
[image]

DSheirer, 2017-07-16:
I've been running virtually non-stop for 7 days and haven't been able to reproduce the TEARDOWN lockup so far.  I know we have encrypted talkgroups on the local system for the counter-drug task force - I'll keep monitoring to see if I can make the lockup happen.

chudgoo, 2017-07-17:
Man, that's impressive. I just got to work and fired up the (#3) test setup of sdrtrunk, my most powerful system by far, and within 10 minutes I had the control channel go to 'OVERFLOW' and just stayed there. No 'TEARDOWN' message. I then went to the Channels tab and tried to disable that control channel.  Waterfall stopped, full GUI lockup, but still heard one last garbled voice call.

chudgoo, 2017-07-17:
So just completed running a few tests (while screen recording, so I wouldn't miss anything), and it appears that it is most easy to replicate when more than one control channel is enabled at once. 

With four CCs enabled, it locked up (teardown stuck on the active (broadcasting) control channel) within 40 seconds. It starts with normal "CONTROL", then changed to "FADE" for 3 seconds, and then after 5 seconds of "IDLE", it goes to "OVERFLOW".  

I re-ran the test but with only one CC enabled and didn't encounter it once in over 30 min.  Should I upload the screen recording? Would that help at all? The terminal window and sdrtrunk are both visible.

chudgoo, 2017-07-17:
Crud. I spoke too soon.  
One CC enabled.
[image]
[image]

DSheirer, 2017-07-18:
Were there any errors in the log indicating that the program ran out of memory, when it locked up?

chudgoo, 2017-07-19:
Nothing too unusual I guess?   Sorry to be posting a pic of text, but I had a screen recording going, and closed the terminal before copying the text.
[image]

closed 2017-07-22 by DSheirer

DSheirer, 2017-07-22:
@chudgoo I just released beta 13 with a fix that I hope resolves this issue.  Although I couldn't duplicate on my system the situation that you were experiencing, I highly suspect that it was a threading issue.  I removed the multi-threaded aspect of the channel state processing module and made it single-threaded.  

Please test Beta 13 if/when you have a chance and let me know if you are still experiencing the stuck TEARDOWN state issue?

PR #255 (2017-07-22): 249 Channel TEARDOWN/OVERFLOW state lockup
Resolves #249 - removes the state monitor runnable from the channel state and replaces it with a processing chain heartbeat that is generated by the sample source.

Previously, the state monitor was running on a separate thread and this was causing multi-threading issues.  With this update, all channel state operations (and all module operations) occur on the source's buffer processing thread.
commit be62ae04 (2017-07-22): Resolves #249 - removes the state monitor runnable from the channel
state and replaces it with a processing chain heartbeat that is generated
by the sample source.

Previously, the state monitor was running on a separate thread and this
was causing multi-threading issues.  With this update, all channel state
operations (and all module operations) occur on the source's buffer
processing thread.