DSheirer/sdrtrunk #254: broadcastify streaming

rcummings14, 2017-07-10:
will not connect to broadcastify mountpoint

chudgoo, 2017-07-10:
Stupid question, but have you re-checked the mountpoint/password stuff to make sure it's 100% right?
I seem to recall having an issue a few months ago where the extra space character at the end that I pasted in caused a problem.

rcummings14, 2017-07-10:
I had to remove the shoutcast streaming option in all of my aliases for
some reason then it let me connect to broadcastify. Then I was able to go
back and add all of them again.

rcummings14, 2017-07-10:
Software is awesome by the way  I provide the stream to broadcastify for my
area and I have a shoutcast station I use just to monitor law enforcement.
I think it is very cool you can split your talkgroups between different
streaming services and only have to use one program for everything.

chudgoo, 2017-07-10:
Totally agree. I found my old laptop that I used to use for my P25 streaming last week and holy crap.... Unitrunker, as nice as it might be, is a PITA and the string of things needed were excessive.  Unitrunker>DSD+>VAC>BUTT.... and NO simulcast support... so I also had directional antennas and shielding to prevent the second tower from interfering.   SDRTrunk is SOOOOOOO much nicer....and compact...and portable...and...and...and...

Also, yeah that's a really interesting idea... separate broadcasts into discrete feeds.... hmmmm

DSheirer, 2017-07-10:
@rcummings14 Ray, is your streaming to broadcastify working ok now?

rcummings14, 2017-07-10:
Yes but volume is a little low can't figure out to to turn it up

chudgoo, 2017-07-11:
That reminds me... This idea came up a few months ago...
@DSheirer , did you ever get around to looking into audio post-processing effects?  (normalization, EQ, compressor, limiter, etc)

DSheirer, 2017-07-11:
@chudgoo not yet -- there's an issue open for it #133

closed 2023-11-15 by DSheirer