DSheirer/sdrtrunk #256: Tracking Active Trunked P25 Control Channel

davidcomtois, 2017-07-23:
The problem, the active P25 control channel moves between frequencies for the same system at the same site occasionally over time.

I tried adding 4 'channels' to sdrtrunk to watch each of the more actively used frequencies for the active control channel but it could be any one of the frequencies of the group (about 8). Also, when a transmission comes over one of the three other 'channels' that I'm watching for the control channel, it receives it on both that 'channel' and the extra 'channel' temporarily added by the active control channel itself; resulting in a duplicate recording and stream entry.

Short of having something else 'scan' between the frequencies and locking onto one until it disappears and feeding that audio feed via sound card into sdrtrunk, what am I missing? Is it possible within sdrtrunk itself? I'm investigating how to do what I said just above in Ubuntu but I'm having little luck due to a lack of easily accessible 'virtual audio cables'.

In the future, could sdrtrunk see the extra duplicate entry and choose one or the other if both the same source and target are active at the same time? Or, could you add multiple frequencies for it to scan through per 'channel'?

Alleg-Gitter, 2017-08-24:
dup #61

DSheirer, 2019-01-21:
Implemented with #61

closed 2019-01-21 by DSheirer