DSheirer/sdrtrunk #260: Broadcastify/Icecast streaming doesn't reconnect

DSheirer, 2017-07-30:
Occasionally when a Broadcastify streaming session is connected and drops out, it fails to reconnect and remains in a 'Connecting' state until the application is restarted, or the stream is disabled and then reenabled.

Set the connection attempt to timeout after a period of time so that a new connect attempt can occur.

rcummings14, 2017-08-09:
any luck with getting this fixed?

DSheirer, 2017-08-10:
I think I have a fix.  When there was an error due to a broken connection, I was disconnecting and then immediately reconnecting which was causing it to never reconnect.  

I updated the code to wait a short period of time before trying to reconnect and am testing it now.  So far I've had 2 dropouts today and both reconnected successfully.

rcummings14, 2017-08-10:
Awesome. I've been running it none stop. I have 2 steams going to
broadcastify and hopefully 3 by tomorrow. But not had any major problems
except for the reconnecting issue. I ordered some more sdr receivers  and
 have done away with discriminator tap feeding my sound card. So that is no
longer an issue for me. Im kinda new at this but if there is anything I can
do to help. I will be happy to do it. I will try to send a donation to help
the cause when I am able. Thanks again for all you hard work

DSheirer, 2017-08-16:
@rcummings14 So far the fix looks good.  Running for 5 days straight with automatic reconnects.

closed 2017-08-20 by DSheirer