DSheirer/sdrtrunk #261: Odd feature request... Repeat longer calls when nothing else is coming in.

chudgoo, 2017-08-02:
I've noticed that my local PS radio traffic is typically short burst, 90 second response, 5 min of silence.... rinse repeat.  I was wondering if it would be possible (or make any sense) to allow user configurable repeats of longer calls when no other new audio is coming in. Something like a call length based priority system. 0.7 second "Copy" calls get lower priority than the 90 sec 'what's actually going on' calls.

On this system there are tons of "copy" type calls. Occasionally there are much longer and informative calls that it would be nice to hear twice, provided there isn't another call stacking up in front of it.

Does this make sense? Brutal honesty is welcomed. ;)

DSheirer, 2017-08-02:
That does seem like an odd request :-)

I've been wanting to do a recordings tab that lists all of the recorded calls and supports click to play right within the window.  Maybe I could enhance that with some sort of age-off logic to keep the recordings directory from filling up.  But, you'd be able to find the longer calls in the list and replay them on demand.

chudgoo, 2017-08-02:
Ah! In the context of live streaming though, how would that work?

chudgoo, 2017-08-02:
I suppose that was the part I accidentally left out. I was actually concerned with the live stream aspect. I've found that it's been kind of handy to have calls repeated sometimes... like if you miss the first couple of seconds while monitoring via the phone broadcasting app on your phone.  (this current, uncontrolled repeated call phenomenon is the result of having multiple control channels selected, I think).

davidmckenzie, 2017-08-02:
That's the beauty of streaming - you can just rewind and it should be in your buffer ;)

chudgoo, 2017-08-02:
But on a Broadcastify feed, going out to phones and everything else, that's not always an option.

chudgoo, 2017-08-02:
Waaaait a minute @davidmckenzie ...  What app are you using to listen to broadcastify streams?  Are there clients that allow more options like transport controls?

davidmckenzie, 2017-08-02:
Yeah I didn't think tha one through - I don't usually listen on the phone ;) I use mplayer on the desktop for it.

Would've thought by now there'd be a streaming app that can do standard media controls... wouldn't iTunes do it natively?