DSheirer/sdrtrunk #275: Icecast streaming

IDEall, 2017-10-09:
I have Icecast 2.4.2 and streaming from sdrtrunk doesn't work. It is connected to the icecast just there's no audio. In the status windows I can see Quequed and Streamed data as "0".
Already testes as an old icecast and icecast2 stream. I do have decoded audio on line out form my sound card

--
INFO  a.b.AudioBroadcaster - [stream] status: Connected

DSheirer, 2017-10-09:
Have you added a broadcast stream to each of the aliases that you want to stream over icecast?

IDEall, 2017-10-10:
yep, I have configured broadcast channel "stream" and I see filtered aliases in the events

DSheirer, 2017-10-11:
Have you added the broadcast channel 'stream' to each individual alias?  If so, could you please send me your playlist_v2.xml file so that I can check to see what might be preventing the audio from streaming?