DSheirer/sdrtrunk #276: Overflow related freeze  (reproducible, w/video)

chudgoo, 2017-10-11:
Here's an (unlisted) youtube video of the issue:  https://www.youtube.com/watch?v=vF6HEKqRvX0

I am running two instances of sdrtrunk 0.3.1-final. 
They have the same (software) configs, but one has four tuners and the other has two.  

With only one P25 control channel enabled, given sufficient time/traffic/activity (not sure which exactly yet), the application will stop tracking the CC and stop producing audio and show the CC and whichever calls were running at the time in "OVERFLOW" mode. The Messages/Events tabs freeze, but the waterfall and FFT continue to scroll.  

If I go into the Channels tab and disable or enable any of the control channels, the application freezes completely. Nothing is clickable, the waterfall/FFT stop and the window is not resize-able, yet it doesn't "crash" exactly by closing the window.

This issue seemed to go away around beta 11 (i think) but seems to have returned.

The logs don't really indicate the issue, but I'll paste them here:
(note that this machine is a 12 core Mac Pro w/ 64GB RAM, and the same thing happens here as it does on the quad core i7 w/ 16GB RAM using the same mix of tuners (rtlsdr v3, NESDR and FlightAware)

$ bash ./run_sdrtrunk_linux.sh 
java version "1.8.0_144"
Java(TM) SE Runtime Environment (build 1.8.0_144-b01)
Java HotSpot(TM) 64-Bit Server VM (build 25.144-b01, mixed mode)
Using Java 8 program options
10:17:30.664 INFO  gui.SDRTrunk - 
10:17:30.667 INFO  gui.SDRTrunk - *******************************************************************
10:17:30.667 INFO  gui.SDRTrunk - **** sdrtrunk: a trunked radio and digital decoding application ***
10:17:30.667 INFO  gui.SDRTrunk - ****  website: https://github.com/dsheirer/sdrtrunk             ***
10:17:30.667 INFO  gui.SDRTrunk - *******************************************************************
10:17:30.667 INFO  gui.SDRTrunk - 
10:17:30.668 INFO  gui.SDRTrunk - Host CPU Cores: 24
10:17:30.668 INFO  gui.SDRTrunk - Host Memory Total: 1073741824
10:17:30.668 INFO  gui.SDRTrunk - Host Memory Max: 17179869184
10:17:30.668 INFO  gui.SDRTrunk - Host Memory Free: 1052935504
10:17:30.682 INFO  util.ThreadPool - Application thread pool created with [24] threads
10:17:30.682 INFO  gui.SDRTrunk - Home path: /Users/chud/SDRTrunk
10:17:30.684 INFO  p.SystemProperties - SystemProperties - loaded [/Users/chud/SDRTrunk/SDRTrunk.properties]
10:17:30.684 INFO  p.SystemProperties - SystemProperties - application properties loaded [/Users/chud/SDRTrunk/SDRTrunk.properties]
10:17:30.691 INFO  settings.SettingsManager - SettingsManager - loading settings file [/Users/chud/SDRTrunk/settings/settings.xml]
10:17:31.045 INFO  source.mixer.MixerManager - loading system mixer devices
	[LOADED]     Input:  Default Audio Device CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: Default Audio Device CHANNELS: [MONO, STEREO]
	[LOADED]     Output: Built-in Output CHANNELS: [MONO, STEREO]
	[LOADED]     Output: DisplayPort CHANNELS: [MONO, STEREO]
	[LOADED]     Output: DisplayPort CHANNELS: [MONO, STEREO]
	[LOADED]     Output: DisplayPort CHANNELS: [MONO, STEREO]
	[LOADED]     Output: Wireless Stereo Headset CHANNELS: [MONO, STEREO]
	[LOADED]     Input:  Wireless Stereo Headset CHANNELS: [MONO]
	[LOADED]     Input:  Soundflower (2ch) CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: Soundflower (2ch) CHANNELS: [MONO, STEREO]
	[LOADED]     Input:  Soundflower (64ch) CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: Soundflower (64ch) CHANNELS: [MONO, STEREO]
	[LOADED]     Output: Instant On Sound Effects CHANNELS: [MONO, STEREO]

10:17:31.046 INFO  s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations
10:17:31.092 INFO  source.tuner.TunerManager - LibUSB API Version: 16777474
10:17:31.093 INFO  source.tuner.TunerManager - LibUSB Version: 1.0.18.10866
10:17:31.093 INFO  source.tuner.TunerManager - discovered [35] attached USB devices
10:17:32.634 INFO  source.tuner.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T RTLSDR-0
10:17:34.164 INFO  source.tuner.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T NESDR-2-1
10:17:35.691 INFO  source.tuner.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T NESDR-1-1
10:17:37.212 INFO  source.tuner.TunerManager - usb device [0BDA:2832] LOADED: RTL2832 SDR/R820T blue-0

**<SNIPPED OUT THE UNKNOWN DEVICE LISTINGS>**

10:17:40.922 INFO  icon.IconManager - loading icons file [/Users/chud/SDRTrunk/settings/icons.xml]
10:17:41.112 INFO  playlist.PlaylistManager - Attempting to load version 2 playlist file [/Users/chud/SDRTrunk/playlist/playlist_v2.xml]
10:17:41.477 INFO  m.d.p.a.P25AudioModule - JMBE audio conversion library [JMBE Audio Conversion Library v0.3.3] successfully loaded - P25 audio will be available
10:17:41.499 INFO  gui.SDRTrunk - starting main application gui
10:18:17.885 INFO  record.RecorderManager - Removing idle recorder [SRC:3 ID:0C1B]
10:20:07.880 INFO  record.RecorderManager - Removing idle recorder [SRC:2 ID:03E9]

These screenshots were taken before attempting to disable the active control channel
[image]

[image]
[image]
[image]

chudgoo, 2017-10-11:
Here's another example.  https://www.youtube.com/watch?v=THQgXXQ8liI

Does some eagle eyed person see what event might be causing this? 
 Is it a certain type of call or event?  The event details ocassionally show "NO TUNER AVAILABLE : PRI:4 PACKET" when there are FOUR physical tuners running.  So that's kind of odd.

The only thing significant about the Fire Dispatch is that it's configured as audio priority 1 and nearly all of the other calls you see are being ignored as they're set to 'do not monitor' and 'non-recordable'.

davidmckenzie, 2017-10-11:
One weird thing there - your control channel is pushed out to the edge of the spectrum a little before each failure. I've had that happen before but it didn't result in this kind of failure. What I did to prevent that is have one tuner set to a lower sample rate so it didn't move around so much, and just enable and disable channels until it lands on the right tuner (major PITA).

Tried running with Java 9, since that's what 0.3.1 was built with?

Edit: Just to note, I'm still on v0.3.0-beta.14

chudgoo, 2017-10-11:
Huh!  I haven't noticed that... but yeah, if I set it to 407.3000MHz, then all possible control channels fit in the 2.4MHz window, but as calls come in, it starts to shift around.  Is this normal behavior? 

Is there a way of locking a given tuner exclusively to CC duties?

Could it be that if call starts and it doesn't end before this tuning shift occurs it would leave the call 'open' once the call's freq is pushed out of range?  Does that make sense?

chudgoo, 2017-10-11:
Ok, so I'm testing this out...  Running at 1.2MHz per tuner lead to an overflow condition in about 3 minutes. That's about how long it took at 2.4MHz.  I'm now running each tuner set at 2.88MHz. The audio decodes are noticibly choppier, but so far it's been running for 10 minutes without an overflow condition occurring.  I'll continue monitoring...

chudgoo, 2017-10-11:
Ooops, sorry @davidmckenzie I read your comment too fast and misunderstood.  Do you just find the tuner on CC duty by looking at the number of channels it's handling?  Also is it normal that one tuner would have up to 4 channels at a time while the others have zero?
[image]

davidmckenzie, 2017-10-11:
Yeah the tuner picking code seems to preference using as few tuners as possible, which is annoying when you've only got a limited amount of actual usable bandwidth.

What I do is a little hard to explain - basically I disable all channels, enable the control channel and see which tuner it lands on, then set that tuner to super low sample rate, then enable all my other channels. It's a bit of a screw around, but it generally results in having one tuner only serving control channel, one tuner for a couple of non-trunked p25 channels, and the other tuner gets used for any voice calls on the trunking system.

chudgoo, 2017-10-12:
So I am trying the limited bandwidth on the 'chosen one' tuner for CC data on the production machine.  It's holding up surprisingly well.  It's been running (very well) for something like 3 hours now.  This same instance would run between 3-10 minutes before hitting the overflow state before. This new approach seems to be working. Thanks @davidmckenzie 

[image]

I think the timing of beta 11 and my local P25 system swapping to a primary CC that was a full 1.5MHz above the usual one might have lead to the confusion (and assumption) that it was something wrong with SDRTrunk.  

**It seems to me that if control channel tracking and the ability to assign a given tuner to a given role (single CC with a 'multiple voice tuner' pool) were added, many of these weird issues would go away.**

BTW,  @DSheirer  -  In no way do I intend anything negative here. As I've _gushed_ before, this is an amazing application and I've been running it on at least two machines continuously for the last 17 months, and experimented on probably 15 other laptops/SFF PCs/SBCs/netbooks/etc over that time.  It's the first thing I check on in the morning and the first thing I check on when I get home from work.  I TRULY appreciate your work on this project. :)

DSheirer, 2017-10-14:
@chudgoo I really appreciate the effort you're putting into tracking down this issue and I want to get it resolved for you. 

I'm researching how to use Java Mission Control to record the state of the threads/objects at the time of channel lockup so that I can try to figure out what is causing the situation.  As soon as I figure out how to use JMC correctly, would you mind recording the lockup and sending me the recording?

DSheirer, 2017-10-15:
@chudgoo Can you please capture a thread dump report of your running sdrtrunk application.  Please use the following steps:

1. Start sdrtrunk application
2. In separate command prompt, type: jvisualvm to launch the Java VisualVM application
3. In left-hand 'Applications' pane, double-click the gui.SDRTrunk (pid ...) entry to connect to the running sdrtrunk application.
4. On right-hand pane that appears, click the 'Threads' tab.
5. On the top-right corner, there is a 'Thread Dump' button.  Once the sdrtrunk channels go into the 'Overflow' state, click the button to create a thread dump.
6. Right-click in the thread-dump report and select all, right-click again to copy, and then paste the output here.

Thanks in advance,
Denny

chudgoo, 2017-10-16:
@DSheirer  I'll get on it now.

chudgoo, 2017-10-16:
Ok, here's the thread dump after overflow freeze.

2017-10-16 11:29:44
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.144-b01 mixed mode):

"RMI TCP Accept-0" #62 daemon prio=9 os_prio=31 tid=0x00007ff9e325e000 nid=0x11007 runnable [0x0000700006394000]
   java.lang.Thread.State: RUNNABLE
        at java.net.PlainSocketImpl.socketAccept(Native Method)
        at java.net.AbstractPlainSocketImpl.accept(AbstractPlainSocketImpl.java:409)
        at java.net.ServerSocket.implAccept(ServerSocket.java:545)
        at java.net.ServerSocket.accept(ServerSocket.java:513)
        at sun.management.jmxremote.LocalRMIServerSocketFactory$1.accept(LocalRMIServerSocketFactory.java:52)
        at sun.rmi.transport.tcp.TCPTransport$AcceptLoop.executeAcceptLoop(TCPTransport.java:400)
        at sun.rmi.transport.tcp.TCPTransport$AcceptLoop.run(TCPTransport.java:372)
        at java.lang.Thread.run(Thread.java:748)

   Locked ownable synchronizers:
        - None

"Attach Listener" #61 daemon prio=9 os_prio=31 tid=0x00007ff9e5ae9000 nid=0x11c23 waiting on condition [0x0000000000000000]
   java.lang.Thread.State: RUNNABLE

   Locked ownable synchronizers:
        - None

"DestroyJavaVM" #60 prio=5 os_prio=31 tid=0x00007ff9e74e8000 nid=0x1c03 waiting on condition [0x0000000000000000]
   java.lang.Thread.State: RUNNABLE

   Locked ownable synchronizers:
        - None

"TimerQueue" #58 daemon prio=5 os_prio=31 tid=0x00007ff9e6368800 nid=0x1207 waiting on condition [0x0000700006291000]
   java.lang.Thread.State: WAITING (parking)
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x00000003c6471d70> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
        at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2039)
        at java.util.concurrent.DelayQueue.take(DelayQueue.java:211)
        at javax.swing.TimerQueue.run(TimerQueue.java:174)
        at java.lang.Thread.run(Thread.java:748)

   Locked ownable synchronizers:
        - <0x00000003c6462150> (a java.util.concurrent.locks.ReentrantLock$NonfairSync)

"pool-1-thread-4" #57 prio=5 os_prio=31 tid=0x00007ff9e88bb800 nid=0x11a7b waiting on condition [0x000070000618e000]
   java.lang.Thread.State: TIMED_WAITING (parking)
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x00000003c0392950> (a java.util.concurrent.SynchronousQueue$TransferStack)
        at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
        at java.util.concurrent.SynchronousQueue$TransferStack.awaitFulfill(SynchronousQueue.java:460)
        at java.util.concurrent.SynchronousQueue$TransferStack.transfer(SynchronousQueue.java:362)
        at java.util.concurrent.SynchronousQueue.poll(SynchronousQueue.java:941)
        at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1073)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1134)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)

   Locked ownable synchronizers:
        - None

"pool-1-thread-3" #56 prio=5 os_prio=31 tid=0x00007ff9e8965000 nid=0x11793 waiting on condition [0x000070000608b000]
   java.lang.Thread.State: TIMED_WAITING (parking)
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x00000003c0392950> (a java.util.concurrent.SynchronousQueue$TransferStack)
        at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
        at java.util.concurrent.SynchronousQueue$TransferStack.awaitFulfill(SynchronousQueue.java:460)
        at java.util.concurrent.SynchronousQueue$TransferStack.transfer(SynchronousQueue.java:362)
        at java.util.concurrent.SynchronousQueue.poll(SynchronousQueue.java:941)
        at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1073)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1134)
[... 779 lines cut ...]
"G1 Concurrent Refinement Thread#8" os_prio=31 tid=0x00007ff9e3005800 nid=0x5d03 runnable 

"G1 Concurrent Refinement Thread#9" os_prio=31 tid=0x00007ff9e3005000 nid=0x5b03 runnable 

"G1 Concurrent Refinement Thread#10" os_prio=31 tid=0x00007ff9e3004000 nid=0x5903 runnable 

"G1 Concurrent Refinement Thread#11" os_prio=31 tid=0x00007ff9e7003800 nid=0x5703 runnable 

"G1 Concurrent Refinement Thread#12" os_prio=31 tid=0x00007ff9e7003000 nid=0x5503 runnable 

"G1 Concurrent Refinement Thread#13" os_prio=31 tid=0x00007ff9e7801000 nid=0x5303 runnable 

"G1 Concurrent Refinement Thread#14" os_prio=31 tid=0x00007ff9e7800000 nid=0x5103 runnable 

"G1 Concurrent Refinement Thread#15" os_prio=31 tid=0x00007ff9e3003800 nid=0x4f03 runnable 

"G1 Concurrent Refinement Thread#16" os_prio=31 tid=0x00007ff9e3002800 nid=0x4d03 runnable 

"G1 Concurrent Refinement Thread#17" os_prio=31 tid=0x00007ff9e7002000 nid=0x4b03 runnable 

"G1 Concurrent Refinement Thread#18" os_prio=31 tid=0x00007ff9e7001800 nid=0x4903 runnable 

"VM Periodic Task Thread" os_prio=31 tid=0x00007ff9e4801000 nid=0xad03 waiting on condition 

JNI global references: 2596

chudgoo, 2017-10-16:
Here's another that just happened.

2017-10-16 11:40:30
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.144-b01 mixed mode):

"RMI TCP Accept-0" #62 daemon prio=9 os_prio=31 tid=0x00007ffeffc18000 nid=0x31703 runnable [0x00007000093f8000]
   java.lang.Thread.State: RUNNABLE
        at java.net.PlainSocketImpl.socketAccept(Native Method)
        at java.net.AbstractPlainSocketImpl.accept(AbstractPlainSocketImpl.java:409)
        at java.net.ServerSocket.implAccept(ServerSocket.java:545)
        at java.net.ServerSocket.accept(ServerSocket.java:513)
        at sun.management.jmxremote.LocalRMIServerSocketFactory$1.accept(LocalRMIServerSocketFactory.java:52)
        at sun.rmi.transport.tcp.TCPTransport$AcceptLoop.executeAcceptLoop(TCPTransport.java:400)
        at sun.rmi.transport.tcp.TCPTransport$AcceptLoop.run(TCPTransport.java:372)
        at java.lang.Thread.run(Thread.java:748)

   Locked ownable synchronizers:
        - None

"DestroyJavaVM" #61 prio=5 os_prio=31 tid=0x00007ffefe171000 nid=0x1c03 waiting on condition [0x0000000000000000]
   java.lang.Thread.State: RUNNABLE

   Locked ownable synchronizers:
        - None

"TimerQueue" #60 daemon prio=5 os_prio=31 tid=0x00007ffefacdb800 nid=0x11727 waiting on condition [0x0000700006b53000]
   java.lang.Thread.State: WAITING (parking)
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x00000003d140fe50> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
        at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2039)
        at java.util.concurrent.DelayQueue.take(DelayQueue.java:211)
        at javax.swing.TimerQueue.run(TimerQueue.java:174)
        at java.lang.Thread.run(Thread.java:748)

   Locked ownable synchronizers:
        - <0x00000003d18b6778> (a java.util.concurrent.locks.ReentrantLock$NonfairSync)

"pool-1-thread-4" #59 prio=5 os_prio=31 tid=0x00007ffeff160000 nid=0x11d17 waiting on condition [0x0000700006a50000]
   java.lang.Thread.State: TIMED_WAITING (parking)
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x00000003d14001d8> (a java.util.concurrent.SynchronousQueue$TransferStack)
        at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
        at java.util.concurrent.SynchronousQueue$TransferStack.awaitFulfill(SynchronousQueue.java:460)
        at java.util.concurrent.SynchronousQueue$TransferStack.transfer(SynchronousQueue.java:362)
        at java.util.concurrent.SynchronousQueue.poll(SynchronousQueue.java:941)
        at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1073)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1134)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)

   Locked ownable synchronizers:
        - None

"pool-1-thread-3" #58 prio=5 os_prio=31 tid=0x00007ffeff27d800 nid=0x1307 waiting on condition [0x000070000694d000]
   java.lang.Thread.State: TIMED_WAITING (parking)
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x00000003d14001d8> (a java.util.concurrent.SynchronousQueue$TransferStack)
        at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
        at java.util.concurrent.SynchronousQueue$TransferStack.awaitFulfill(SynchronousQueue.java:460)
        at java.util.concurrent.SynchronousQueue$TransferStack.transfer(SynchronousQueue.java:362)
        at java.util.concurrent.SynchronousQueue.poll(SynchronousQueue.java:941)
        at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1073)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1134)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)

   Locked ownable synchronizers:
        - None

[... 712 lines cut ...]
"G1 Concurrent Refinement Thread#8" os_prio=31 tid=0x00007ffefb00e800 nid=0x5d03 runnable 

"G1 Concurrent Refinement Thread#9" os_prio=31 tid=0x00007ffefb00d800 nid=0x5b03 runnable 

"G1 Concurrent Refinement Thread#10" os_prio=31 tid=0x00007ffefa805800 nid=0x5903 runnable 

"G1 Concurrent Refinement Thread#11" os_prio=31 tid=0x00007ffeff802800 nid=0x5703 runnable 

"G1 Concurrent Refinement Thread#12" os_prio=31 tid=0x00007ffeff801800 nid=0x5503 runnable 

"G1 Concurrent Refinement Thread#13" os_prio=31 tid=0x00007ffeff801000 nid=0x5303 runnable 

"G1 Concurrent Refinement Thread#14" os_prio=31 tid=0x00007ffeff800000 nid=0x5103 runnable 

"G1 Concurrent Refinement Thread#15" os_prio=31 tid=0x00007ffefa804800 nid=0x4f03 runnable 

"G1 Concurrent Refinement Thread#16" os_prio=31 tid=0x00007ffefa804000 nid=0x4d03 runnable 

"G1 Concurrent Refinement Thread#17" os_prio=31 tid=0x00007ffeff001800 nid=0x4b03 runnable 

"G1 Concurrent Refinement Thread#18" os_prio=31 tid=0x00007ffeff001000 nid=0x4903 runnable 

"VM Periodic Task Thread" os_prio=31 tid=0x00007fff00041800 nid=0xad03 waiting on condition 

JNI global references: 1596

DSheirer, 2017-10-16:
Ugh!  Unfortunately, it's what I suspected ... dead-locked threads.  On the bright side, now I know what to fix.

Thanks very much @chudgoo !!

DSheirer, 2017-10-21:
Issue was tracked down to the TunerController.updateLOFrequency() method where it would enter an infinite loop while trying to find a center tune frequency when the frequency separation between the highest channel and a center channel that overlaps the central DC spike avoid zone was exactly 1/2 of the tuner's bandwidth. Fix will be merged into master and then cherry-picked onto a 0.3.2 development branch for beta testing.

DSheirer, 2017-10-21:
@chudgoo I published a V.0.3.2-beta1 release to patch this issue.  Please test when you have an opportunity and let me know if the issue is resolved.  I'll keep this ticket open until we confirm that issue is resolved.

chudgoo, 2017-10-22:
Sorry for the late response. Will install, test now and report back soon.  Thanks @DSheirer

chudgoo, 2017-10-23:
I've been running one dual tuner instance since last night around 10PM and it was still running this morning, though the CC changed on me overnight.  Also, I've been running a two tuner instance since about 7:30 this morning and it's been working flawlessly.  I even have all aliases enabled for extra stress testing.   Whatever you did seems to have fixed the issue for me!  Thanks @DSheirer !

chudgoo, 2017-10-24:
Not to keep flooding this thread, but both instances have been rock solid for the last ~40 hours.  MUCH better than the 3 minutes I was getting before.  ;)

However, the home (production) instance lost network connectivity (thanks comcast) and the streaming tab was stuck on 'connecting' because OSX is dumb sometimes and doesn't automatically reconnect when the wireless gateway power cycles. Going to swap to wired ethernet tonight to help with this snag.  VERY good work @DSheirer  !

DSheirer, 2017-11-04:
Closing issue as resolved.

PR #279 (2017-10-21): 276 TrafficController Infinite Loop
Resolves #276 Traffic Channel threads locking in overflow state.  Issue occured when the ChannelProcessionManager was allocating a new channel
causing the tuner to invoke the updateLOFrequency() method that was
trying to find a center tune frequency that would accommodate all of the
channels, specifically in the instance where the distance between the
highest channel and the middle channel was exactly one-half of the
tuner's bandwidth.  This situation would cause the calculated adjustment
to be zero and would cause an infinite loop, locking all channels from [... cut]
commit f3d3a818 (2017-10-21): Resolves #276 Traffic Channel threads locking in overflow state.  Issue
occured when the ChannelProcessionManager was allocating a new channel
causing the tuner to invoke the updateLOFrequency() method that was
trying to find a center tune frequency that would accommodate all of the
channels, specifically in the instance where the distance between the
highest channel and the middle channel was exactly one-half of the
tuner's bandwidth.  This situation would cause the calculated adjustment
to be zero and would cause an infinite loop, locking all channels from
accessing the any channel startup/t [... cut]
commit 73b6ef16 (2017-10-21): Merge pull request #280 from DSheirer/276-tuner-controller-issue-cherry-pick

Resolves #276 Traffic Channel threads locking in overflow state.  Issue
commit b2c7ac8c (2017-10-21): Resolves #276 Traffic Channel threads locking in overflow state.  Issue
occured when the ChannelProcessionManager was allocating a new channel
causing the tuner to invoke the updateLOFrequency() method that was
trying to find a center tune frequency that would accommodate all of the
channels, specifically in the instance where the distance between the
highest channel and the middle channel was exactly one-half of the
tuner's bandwidth.  This situation would cause the calculated adjustment
to be zero and would cause an infinite loop, locking all channels from
accessing the any channel startup/t [... cut]
closed 2017-11-04 by DSheirer