DSheirer/sdrtrunk #278: Add metadata to WAV file.

DominicWeston, 2017-10-17:
Suggestion: Add metadata to WAV file header. Add a NULL delimited string to the header of the wave file to allow play back programs to display the info and some file browsers too. The HomePatrol scanner included this data in the WAV's header area. What follows is part of a discussion six years ago about the data included in the WAV file, it may be a little confusing but if you imagines "some" of the spaces are actually NULL characters......

The file 2011-07-17_19-45-06.wav shows this in its header:

RIFF WAVELISTp INFOIART@ Howard County IGNR@ Police scue - Administration INAM@ Southern District Dispatch 1 ons) channels A16/E16/F16/G16) ICMT@ 47152 IPRD HomePatrol IKEY , � & ICRD 20110717194508 fmt @ �> data� 

Breaking it down it tells me:

RIFF WAVE                                   Its a wave file, the file size is also here.
LISTp                                            Not sure, I will compare this to an out of state file someday to see if its the state.
INFOIART@ Howard County        Yep, thats obviously the county
IGNR@ Police scue - Administration           THIS IS POLICE, thats a NULL char between PoliceNULLscue which tells me the file before this one was 'Fire/Rescue - Administration and that the HP1 does not clear the string before it reuses it. SO LOOK FOR THE NULL TERMINATION.

INAM@ Southern District Dispatch 1 ons) channels A16/E16/F16/G16)     Yep its Howard County Police Southern District Dispatch 1 and some leftover from the prior string.
^NULL 
ICMT@ 47152                   And the trunk channel is 47152
IPRD HomePatrol              Made on the HP
IKEY , � &                        Don't know
ICRD 20110717194508     Date/time with 2 more seconds on it than the file name.
fmt @ �>                          The format of the file which is uncompressed PulseCodeModulation, monophonic, 128kbps, 
data�                                Says this is the start of the audio bits

DominicWeston, 2017-10-23:
Let me expand on the above and why we would want it. I wrote a program that allowed me to playback audio from when I was away. It played tracks back in the order recorded. What I am adding to it is the ability to place/pan certain signals across the right/left speakers. Say dispatch in the center speaker and police on right and EMS on left and others in between. The meta data we have would be good along with both TG id's and the text of the icon too. I can flesh this out more if there is any interest.

DSheirer, 2017-10-23:
Dominic, I can add support for this to the wav files.  Let me know what metadata you want?

-Frequency/Channel
-System/Site/Channel name
-IDs (radio ID / talkgroup)
-etc.

DominicWeston, 2017-10-23:
I will put that together and append it to the change request and cc you.
 Thanks
   Dom

DominicWeston, 2017-10-23:
Ok, I think I put everything in a logical order. If the wave file was sent through "strings" it would come out in a logical to a human order. Each line is null terminated. Comma separated could be too long?

Date/time 20171023174512345
System
Site
Alias list title here if wanted?
Primary from
Primary from TGID in hex or decimal?
Primary fromRadioID
Primary from Freq if wanted----Could make for unique channel gui?
Secondary from
Secondary from TGID in hex or decimal?
Secondary fromRadioID
Secondary from Freq if wanted----Could make for unique channel gui?
Icon text for simple playback sort.

Fill free to rename, reorder or remove anything as I am still getting my head around what from RadioRef goes where into SDRTrunk.

DominicWeston, 2017-10-23:
Ok, I think I put everything in a logical order. If the wave file was sent through "strings" it would come out in a logical to a human order. Each line is null terminated. Comma separated could be too long?

Date/time 20171023174512345
System
Site
Alias list title here if wanted?
Primary from
Primary from TGID in hex or decimal?
Primary fromRadioID
Primary from Freq if wanted----Could make for unique channel gui?
Secondary from
Secondary from TGID in hex or decimal?
Secondary fromRadioID
Secondary from Freq if wanted----Could make for unique channel gui?
Icon text for simple playback sort.

Fill free to rename, reorder or remove anything as I am still getting my head around what from RadioRef goes where into SDRTrunk.
  Dom

On Oct 23, 2017, at 8:04 AM, Dominic H Weston <dhweston@hotmail.com<mailto:dhweston@hotmail.com>> wrote:

I will put that together and append it to the change request and cc you.
 Thanks
   Dom

DominicWeston, 2017-11-02:
I have gotten the Processing GUI working and testing goes good on HomePatrol recordings. I have a Netbeans GUI and am learning Java to port the Processing/Java to pure Java. There have been no comments on the file format posted above so that should be good then.
  Dom

DominicWeston, 2017-11-03:
I wonder if the idea of using audio positioning in SDRTrunk would work?

chudgoo, 2017-11-03:
(pasting from the FB group for visibility/more people to poke holes in the idea.  ;)

I wonder if you couldn't use this meta data for more... Maybe some type of 'event correlation'....like the ability to make a 'timeline' view per call out. For example, fire dispatch puts out a call saying 'engine 1 and 2 to respond to some address on a report of blah".... then then let's say they assign the call to TAC-2. The following calls would be seen on TAC-2. The initial dispatch call and subsequent calls on TAC-2 would be assigned as an 'event' that you could play start to finish. Something like that... An example of the logic might be "Fire dispatch call makes new event. The fire TAC calls coming within X seconds of that call are lumped in with the dispatch 'initiator' call event. Does that make sense?

DominicWeston, 2017-11-06:
Punching holes is good. I figure there are three main sources of info for SDRTrunk, the news group, the Github pages and FB. I figured the FB and Github would be less of a hijacking and this could get its own Github page at some point. If it becomes popular enough I would hope it stays on the FB page with SDRTrunk as a useful add-on tool. Only problem with a timeline might be is if police are also dispatched to a medical scene you could/would miss that.

DominicWeston, 2017-11-10:
I would add lat/long if someone ever wants to add a map to see what happened during playback.
I have got the Processing program running and ready to test.Rewrite from Processing to Java may take a while.

DominicWeston, 2017-11-18:
Denny

Right now I have my code to show the "Time", "Icon Text", "To", "From", "Sec. To", "Sec. From", "System", "Site", "Name", "Channel", “Frequency” much like the “Now Playing” tab. Program reads null terminated strings and is ready to test/modify when the metadata is ready. If wave files can hold a comma separated string then it may be simpler to go that route.
Dom
FYI   Wav wiki metadata info  2 Metadata<https://en.wikipedia.org/wiki/WAV#Metadata>

On Oct 23, 2017, at 6:00 PM, Dominic H Weston <dhweston@hotmail.com<mailto:dhweston@hotmail.com>> wrote:

Ok, I think I put everything in a logical order. If the wave file was sent through "strings" it would come out in a logical to a human order. Each line is null terminated. Comma separated could be too long?

Date/time 20171023174512345
System
Site
Alias list title here if wanted?
Primary from
Primary from TGID in hex or decimal?
Primary fromRadioID
Primary from Freq if wanted----Could make for unique channel gui?
Secondary from
Secondary from TGID in hex or decimal?
Secondary fromRadioID
Secondary from Freq if wanted----Could make for unique channel gui?
Icon text for simple playback sort.

Fill free to rename, reorder or remove anything as I am still getting my head around what from RadioRef goes where into SDRTrunk.
  Dom

On Oct 23, 2017, at 8:04 AM, Dominic H Weston <dhweston@hotmail.com<mailto:dhweston@hotmail.com>> wrote:

I will put that together and append it to the change request and cc you.
 Thanks
   Dom

DSheirer, 2017-12-22:
Wave metadata tag names: (RIFF INFO tags)
https://sno.phy.queensu.ca/~phil/exiftool/TagNames/RIFF.html#Info

DSheirer, 2017-12-25:
@DominicWeston please download version 0.3.3-beta 1 and test the new audio metadata enhancment.

Please let me know if everything looks ok or if you need something changed?

Note: due to the way that sdrtrunk builds up the collective set of metadata over the life-cycle of a call, I had to add the metadata LIST and ID3 chunks to the end of the recording (versus the way that home patrol adds it at the beginning).  This approach also seems to make the metadata more compatible with a number of playback tools.

I'll keep this ticket open until you indicate that the metadata enhancements are ok.

Denny

closed 2018-02-10 by DSheirer