DSheirer/sdrtrunk #281: Allow RTL-SDR Tuner via TCP

HoracioDos, 2017-10-21:
Hello!
Would you please consider to allow connecting RTL-SDR Tuner via TCP with RTL_TCP command or AirSpy Server.
Thanks!

chudgoo, 2017-10-26:
Wouldn't that only work for fixed frequencies?  
Wouldn't code be required to remotely retune the tuner on the fly?

NoviceLinux-CodingGeek, 2017-10-27:
Probably wouldn't be able problem for the control channel.

classicfm, 2017-12-01:
@chudgoo No, GQRX changes frequencies and other settings (e.g. LNA) on the fly. The client → server link is used to remotely control the dongle.

This would be super useful. As it is, I have to run SDRTrunk by X11 or VNC to get around this, and even over a gigabit Ethernet link, X11 is ... not ideal. That's to say nothing of the problem of audio forwarding over the network. The latency of X11 is probably the fault of AWT and not SDRTrunk, but supporting remote SDR protocols would be very nice.

chudgoo, 2017-12-02:
@classicfm What is the typical network bandwidth used for a remote RTL-SDR sending IQ data over the network?  So if let's say tcp port 3333 is being used to send data, how does the tuning config actually change?  I've only ever used rtl_tcp for one freq at a time.

classicfm, 2017-12-02:
@chudgoo It varies on sampling rate but it's no different than PCM audio. Instead of 16 bit samples the high eight bits are amplitude and the low are phase. Obviously sampling rates are higher than PCM audio but I would say it's about two to three times more bandwidth than CD quality audio.

It easily works over a WiFi link and it even works, albeit rather shakily, over a VPN from my home network (5-6 Mbps upload). In that case it tends to fail because most SDR applications are intolerant of latency changes. 

The control commands are sent over the TCP link. The sample data comes down one side of the bidirectional duplex link, and any configuration commands are sent on the 'reverse' link. I don't know how the application handles e.g. a change in frequency as a consequence of setting it at runtime, I assume it is via internal state tracking.

skawikk, 2020-03-22:
@DSheirer Is there any chance to add RTL-SDR TCP support?

DSheirer, 2020-03-23:
There is a chance, but right now I have so many other features in the queue that I have to get done first.

skawikk, 2020-07-20:
@DSheirer Hi;)
You're doing awesome job!!
Can You post any update in this topic?
M.

Brad-git-man, 2021-06-12:
I'd still really like to see squelch for NFM, that would be great. At the present time in ver 0.6 there is no way to use non-constant carrier NFM frequencies due to the program having a lack of squelch.

ge0metrix, 2022-08-13:
+1 to this, another use case of having my dongles all plugged into my PI near the window and using the horsepower of my desktop to run SDRTrunk.

sofakng, 2022-09-27:
I'm also really hoping to see this!

ArrEssJay, 2023-06-20:
Likewise, my rtl-sdr dongle is connected to a remote headless pi in a different building running rtl_tcp. Bit of a deal breaker not being able to access it remotely.

theficus, 2023-08-05:
Like others I have my SDRs plugged into a Raspberry Pi that's remote. I'm able to use trunk-recorder today to record trunked systems, but it would be nice to take advantage of some of the advanced features that SDRTrunk has.

Peterkal2112, 2023-11-22:
Any Update ? this should be priority must have feature.

Or at least recommend some workaround...

suzieq-1, 2024-08-11:
Hi, 
Until this is added it's not usable for all users, i use spyserver with my airspy and would like to use your software, but i can't.
Also RTL_TCP is not implemented so even "the good old method" doesn't work.
My station is 2000km away, so i can't use it.
Please add this feature.

zach6439, 2024-12-29:
I would also love a native way to incorporate remote tuners. 

Found this channel in my research, he uses remote USB to accomplish this task.
https://www.youtube.com/watch?v=9RAR1lo_Yc0&list=WL&index=2

wbwarnerb, 2025-07-27:
As an alternative, could we get sdrtrunk to run in a headless mode on a server and run a cheap/simple web UI?

pilot51, 2025-07-27:
It automatically runs headless if there's no windowing system. I've been running it in a tmux session on my headless server.

A web UI would be very much appreciated. That should be a new issue. Related: #212

wbwarnerb, 2025-07-30:
> It automatically runs headless if there's no windowing system. I've been running it in a tmux session on my headless server.
> 
> A web UI would be very much appreciated. That should be a new issue. Related: [#212](https://github.com/DSheirer/sdrtrunk/issues/212)

Thanks for the info.  I'm not seeing any documentation on headless operation... but yes, as you mentioned when run on a server it reports it is running headless.  But... what is it actually doing?  I don't see a port bind.  How is the headless session managed? Is there a client that connects to it?

wbwarnerb, 2025-07-31:
> It behaves exactly the same as it does with a UI, except without the UI to control or monitor it. As far as I'm aware, there is no built-in remote access other than streaming if you have that set up.
> 
> I copied the config files from the installation on my main PC and edited them to save recordings to a different directory on the server and use the correct receivers for each channel. I use SFTP in my file browser (Dolphin) to check the recordings, usually when I notice something curious going on like emergency vehicles or the road commission working on my street.
> 
> I had to `export JAVA_OPTS="-Xmx4G"` to prevent an `OutOfMemoryException`. It defaulted to 1GB and frequently ran out of memory (hours or days) and I had to `kill -9`. It still sometimes runs out of memory at 4GB, but much less often (weeks or months). I suspect a memory leak. I'm not going higher than 4GB since the server (Intel NUC 5 from 2016) only has 8GB RAM.

That's pretty smart. I didn't understand it auto records. I'll revisit the configs.  You should do a medium/blog post on your setup. 

If your server is on a private IP and your'e comfortable with it, you can run a python http.server ("python3 -m http.server [port]") from the recordings folder, it should just render the files in the folder to the browser, to download/play/etc.

szechyjs, 2026-09-16:
I have this working, I'll push a PR soon

szechyjs, 2026-09-24:
Created PR #2471

DSheirer, 2026-11-15:
PR 2471 is merged.

closed 2026-11-15 by DSheirer